ArduCAM/Arduino #438: ArduCAM mini 2MP not working with LS 8020 lens

anku2424, 2019-08-19:
Hello Team ArduCAM!
Recently i bought ArduCAM 2MP mini camera board and it works great but it doesn't have ir sensitive lens. So i bought another lens LS 8020 i.e. ir sensitive but i am not able to get images with it. I mean the images that it is taking is a mixture of colours like the one i've shown below.
[image]
Is it possible to use this lens with ArduCAM mini 2MP or 5MP board? If not, what other lens can i use to make it ir sensitive?

UCTRONICS, 2019-08-19:
@anku2424 
It seems the focus is not ok. First adjust the focus manually.

anku2424, 2019-08-19:
After manually adjusting the focus through host app continuous shots, the image is a little better but still it is not satisfactory. It is still blurry. What can i do here?