ArduCAM/Arduino #442: Help porting to Arduino Nano BLE Sense

dansitu, 2019-09-10:
Hi! I'm attempting to use the ArduCam Mini 2MP Plus with the new [Arduino Nano BLE Sense](https://store.arduino.cc/usa/nano-33-ble-sense-with-headers) board. To get started, I've been trying to run the `ArduCAM_Mini_2MP_Plus_functions` example.

The build initially fails since there's no set of `#define` statements for the board in `ArduCam.h`. I've added the section in **Snippet 1** below (the board's identifying macro is `NRF52840_XXA`). The defines were copied from the `__SAM3X8E__` section, but with anything that was already defined removed.

However, with this added, the build still fails with a message that the `portOutputRegister` and `digitalPinToBitMask` macros, which are used in `ArduCam.cpp` and are shown in **Snippet 2** below, are undefined - I suppose these functions are not available for this board. To try and work around this, I added `NRF52840_XXA` to the list of exceptions, as in **Snippet 3** below. At this point, the sample builds, but `ACK CMD SPI interface Error! END` is logged when the program runs.

Can you provide any guidance for getting the device working on this board? If we can get it working in the next week or so, I'll be recommending the ArduCam in an [upcoming book on embedded machine learning](http://shop.oreilly.com/product/0636920254508.do).

Warmly,
Dan

**Snippet 1 - ArduCam.h**
```c++
#if defined(NRF52840_XXAA)

#define cbi(reg, bitmask) *reg &= ~bitmask
#define sbi(reg, bitmask) *reg |= bitmask

#define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
#define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);

#define cport(port, data) port &= data
#define sport(port, data) port |= data

#define swap(type, i, j) {type t = i; i = j; j = t;}
#define fontbyte(x) cfont.font[x]  

#define regtype volatile uint32_t
#define regsize uint32_t

#define PROGMEM

#if defined F
	#undef F
#endif
#define F(X) (X)
#endif
```

**Snippet 2 - ArduCam.cpp**
```c++
		#if (defined(ESP8266)||defined(ESP32)||defined(TEENSYDUINO))
		  B_CS = CS;
		#else
		  P_CS  = portOutputRegister(digitalPinToPort(CS));
		  B_CS  = digitalPinToBitMask(CS);
		#endif
```

**Snippet 2 - ArduCam.cpp with modification**
```c++
		#if (defined(ESP8266)||defined(ESP32)||defined(TEENSYDUINO)||defined(NRF52840_XXAA))
		  B_CS = CS;
		#else
		  P_CS  = portOutputRegister(digitalPinToPort(CS));
		  B_CS  = digitalPinToBitMask(CS);
		#endif
```

dansitu, 2019-09-10:
Update on this: the Arduino team recommended the following patch. However, I'm still seeing `ACK CMD SPI interface Error! END`.

```
diff --git a/ArduCAM/ArduCAM.cpp b/ArduCAM/ArduCAM.cpp
index 0fc338f..84c67a5 100644
--- a/ArduCAM/ArduCAM.cpp
+++ b/ArduCAM/ArduCAM.cpp
@@ -130,7 +130,7 @@ ArduCAM::ArduCAM(byte model ,int CS)
                        B_CS = CS;
                }
        #else
-               #if (defined(ESP8266)||defined(ESP32)||defined(TEENSYDUINO))
+               #if (defined(ESP8266)||defined(ESP32)||defined(TEENSYDUINO)||defined(ARDUINO_ARCH_MBED))
                  B_CS = CS;
                #else
                  P_CS  = portOutputRegister(digitalPinToPort(CS));
diff --git a/ArduCAM/ArduCAM.h b/ArduCAM/ArduCAM.h
index 9542351..db434e1 100644
--- a/ArduCAM/ArduCAM.h
+++ b/ArduCAM/ArduCAM.h
@@ -234,7 +234,7 @@
        #define F(X) (X)
 #endif
 
-#if defined(ARDUINO_ARCH_NRF52)
+#if defined(ARDUINO_ARCH_NRF52) || defined(ARDUINO_ARCH_MBED)
     #define cbi(reg, bitmask) digitalWrite(bitmask, LOW)
        #define sbi(reg, bitmask) digitalWrite(bitmask, HIGH)
        #define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
```

UCTRONICS, 2019-09-17:
@dansitu 
Thank you for choosing Arducam.  Don't worry and I will try my best to help you.
We will buy the Arduino Nano BLE and debug it on this platform.

dansitu, 2019-09-17:
Fantastic, thank you! If we can figure this out by 10/7, we'll be able to use the camera module in our book :)

UCTRONICS, 2019-09-23:
@dansitu  
Hello， I have optimized the ArduCAM library and add support the nano ble platform.
Please try replace the new ArduCAM.cpp and ArduCAM.h file or download our new ArduCAM library.
[image]
[image]
[image]

dansitu, 2019-09-25:
Thank you very much, I will try this today!

dansitu, 2019-10-07:
It works, thank you :)

closed 2019-10-07 by dansitu

UCTRONICS, 2019-10-08:
@dansitu 
You are welcome, Great to hear your success. Looking forward to your new  upcoming book on embedded machine learning.

dansitu, 2019-10-08:
Thank you! In case anyone is interested, here's the book:

[TinyML: Machine Learning with TensorFlow on Arduino, and Ultra-Low Power Micro-Controllers](http://shop.oreilly.com/product/0636920254508.do)

I'll also share the link to our example code featuring Arducam when it is open-sourced later this week!

Dayananda-V, 2020-07-08:
Hi @UCTRONICS 

Which program u use to stream camera using nano ble 33 sense h/w? Is possible to share demo program source code?

rodolphekoe, 2021-11-17:
Hello,
It doesn't work with Arduino Nano RP2040 connect.
Our camera is the ArduCam mini 5MP OV5642 Plus, we configured the memorysaver.h as explained in the tutorial.
In fact, we tried to compile the example ArduCAM_Mini_5MP_OV5642_Plus_Functions, and we get this error message :

Arduino : 1.8.16 (Mac OS X), Carte : "Arduino Nano RP2040 Connect"

In file included from /Users/rodolphek/Documents/Arduino/libraries/ArduCAM/examples/mini/ArduCAM_Mini_5MP_OV5642_Plus_Functions/ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino:13:0:
/Users/rodolphek/Documents/Arduino/libraries/ArduCAM/ArduCAM.h:768:2: error: 'regtype' does not name a type; did you mean 'wctype'?
  regtype *P_CS;
  ^~~~~~~
  wctype
/Users/rodolphek/Documents/Arduino/libraries/ArduCAM/ArduCAM.h:769:2: error: 'regsize' does not name a type
  regsize B_CS;
  ^~~~~~~
/Users/rodolphek/Documents/Arduino/libraries/ArduCAM/examples/mini/ArduCAM_Mini_5MP_OV5642_Plus_Functions/ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino: In function 'void loop()':
/Users/rodolphek/Documents/Arduino/libraries/ArduCAM/examples/mini/ArduCAM_Mini_5MP_OV5642_Plus_Functions/ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino:763:19: warning: unused variable 'temp_last' [-Wunused-variable]
     uint8_t temp, temp_last;
                   ^~~~~~~~~
exit status 1
Erreur de compilation pour la carte Arduino Nano RP2040 Connect

Can you provide any guidance for getting the device working on this board? 
Thank you ! 👍 🐙