ArduCAM/Arduino #446: ArduCAM Mini 2 MP Not sending photo

angeac, 2019-10-02:
Hi, I'm working on a project where I need to take some photos, Im using ArduCAM mini 2 mp and I'm following this tutorial: https://www.youtube.com/watch?v=hybQpjwJ4aA the issue is that when I use the app and click on capture I dont recieve the photo. The monitor says " ArduCAM Start! END",  " SPI interface OK. END" and " OV2640 detected. END", when I click it says " CAM start single shoot. END" but doesnt send the picture. I see the arduino and the TX light neither blink (I supose it has to). Please help, I'm working on my degree's project.

UCTRONICS, 2019-10-02:
@angeac 
Don't worry and I will try my best to help you.
Please attach me your hardware setup and I will help you check it in detail

angeac, 2019-10-02:
Using Arduino Uno adn ArduCAM-Mini-2MP, all the 8 pins are conected correctly. 
This is what im geting. Im using the example named "ArduCAM_Mini_2MP_OV2640_functions"

[image]

angeac, 2019-10-02:
[Code.txt](https://github.com/ArduCAM/Arduino/files/3683645/Code.txt)

angeac, 2019-10-02:
// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can test OV2640 functions
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP platform.
#if !(defined OV2640_MINI_2MP)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
#if defined (OV2640_MINI_2MP)
  ArduCAM myCAM( OV2640, CS );
#else
  ArduCAM myCAM( OV5642, CS );
#endif
uint8_t read_fifo_burst(ArduCAM myCAM);
void setup() {
// put your setup code here, to run once:
uint8_t vid, pid;
uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
  Serial.begin(115200);
#else
  Wire.begin();
  Serial.begin(921600);
#endif
Serial.println(F("ACK CMD ArduCAM Start! END"));
// set the CS as an output:
pinMode(CS, OUTPUT);
digitalWrite(CS, HIGH);
// initialize SPI:
SPI.begin();
  //Reset the CPLD
myCAM.write_reg(0x07, 0x80);
delay(100);
myCAM.write_reg(0x07, 0x00);
delay(100);
while(1){
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55){
    Serial.println(F("ACK CMD SPI interface Error! END"));
    delay(1000);continue;
  }else{
    Serial.println(F("ACK CMD SPI interface OK. END"));break;
  }
}

#if defined (OV2640_MINI_2MP)
  while(1){
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))){
      Serial.println(F("ACK CMD Can't find OV2640 module! END"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("ACK CMD OV2640 detected. END"));break;
    } 
  }
#else
  while(1){
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    if((vid != 0x56) || (pid != 0x42)){
      Serial.println(F("ACK CMD Can't find OV5642 module! END"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("ACK CMD OV5642 detected. END"));break;
    } 
  }
#endif
//Change to JPEG capture mode and initialize the OV5642 module
myCAM.set_format(JPEG);
myCAM.InitCAM();
#if defined (OV2640_MINI_2MP)
  myCAM.OV2640_set_JPEG_size(OV2640_320x240);
#else
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_320x240);
#endif
delay(1000);
myCAM.clear_fifo_flag();
#if !(defined (OV2640_MINI_2MP))
myCAM.write_reg(ARDUCHIP_FRAMES,0x00);
#endif
}
void loop() {
.
.
.

angeac, 2019-10-03:
@UCTRONICS

UCTRONICS, 2019-10-05:
@angeac 
The hardware connection is right.
Please try this demo https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming/ArduCAM_Mini_Video_Streaming.ino

Bokoblin, 2019-11-02:
Hello, I'm also encountering the same issue with the Mini_5MP_Plus model and an Arduino Uno. I have plugged all cables accordingly and tried all the different examples for that specific model with both versions of the host app (v1 and v2).

If I try the examples without the host (so with the Arduino Serial util), each command is always acknowledged the same, followed by broken symbols (see screenshot attached).
[image]

[image]

Is there anything I can do to solve that issue @UCTRONICS?
Did you managed to make it work since your last comment @angeac?

By the way, I first tried to work with an Arduino Mega board (Elegoo 2560 R3) but received an SPI error at setup time.

UCTRONICS, 2019-11-04:
@Bokoblin 
Yes, because the code print the image data with char type. You can open the code and change the Serial.write to Serial.println(), Then you can print it with special type such as hex or DEC.

Bokoblin, 2019-11-04:
> @Bokoblin
> Yes, because the code print the image data with char type. You can open the code and change the Serial.write to Serial.println(), Then you can print it with special type such as hex or DEC.

Okay, thanks, I'll try that for the serial output part.
However, it won't solve the two aforementioned issues (serial input and no picture in host app)

Bokoblin, 2019-11-04:
@UCTRONICS 
I just tried and your solution indeed removes the weird characters and shows scarcely only some groups of A and F hexa chars.

Concerning speaking with the camera, I don't know how your sample is supposed to do it properly.
It has a switch case on some hexa values but when I enter the target value in the serial console, it never reaches the target block in the switch case.
For example, typing **0x40** doesn't let me enter the case block with the line `Serial.println(F("ACK CMD Set to Auto END"));`, instead I obtain:
```
ACK CMD CAM start single shoot. END
ACK CMD CAM start single shoot. END
ACK CMD CAM Capture Done. END
```

UCTRONICS, 2019-11-05:
@Bokoblin 
Let's move on the first question. Do you want to use our Host V2 to preview image and save image to local disk?
If so, you don't need to change anything and just use the videostreaming demo we provide. Just
open the Host V2 and choose the right baud rate and click the open and capture to get one frame image.
You don't need to open the serial monitor, due to the image data will be send to the host V2 with char type. So  if you open the serial monitor, you will just see the broken symbols, which in fact is the image data.

Bokoblin, 2019-11-05:
@UCTRONICS Yeah, I understand that: without touching any of the code and by selecting the proper port and baud value, I see no preview image in the host app, but its log says otherwise ("ArduCam Single, cam start single shoot", exactly like OP October 3rd image)

UCTRONICS, 2019-11-05:
@Bokoblin 
Your project sounds great. Don't worry and I will help you solve your problems.
It seems the Host V2 does not receive the data from the sensor. Please ensure 
The port is not occupied by another application.  If the port is occupied by another application,
It will not work normally.
Our Host V2 support save image and preview image.  Which demo are you using? I will help you test it.

Bokoblin, 2019-11-05:
@UCTRONICS I don't have a direct access to my computer right now, I'll double check the port occupancy, but through previous tests I could see that if it was occupied (by the serial console for example), the host app would not succeed the "open" operation.
I have tried both : ArduCAM_Mini_2MP_Plus_functions
and ArduCAM_Mini_2MP_Plus_VideoStreaming
demos

Bokoblin, 2019-11-06:
@UCTRONICS I have retried. Port does not seem occupied by any other program.
I get the following image on host app by clicking on open then capture.
Weirdly this time there is no printing concerning SPI interface and camera model.
BTW, the serial console was not opened at any moment.

[image]

I also send you a screenshot of COM ports from device manager with the used port (com4) advanced settings (they are in French but if you open the same way the settings on your PC, the settings will correspond).
I see there that the baud rate is set to 9600, but as long as it is correct in the arduino file I'm not sure, that causes any issue. Available rates in device manager, go up to 128000 there anyway, so it can't be changed to match arduino file values I imagine.

[image]

UCTRONICS, 2019-11-07:
@Bokoblin 
Hello,
Have you tested the camera on official Arduino UNO board? 
If your board does not support 921600 baud rate, you can try change the baud rate to 115200 in the code and try again.
I advise you test the camera on official UNO platform firstly to ensure the hardware is normal.

Bokoblin, 2019-11-07:
@UCTRONICS On the official Arduino Uno, 921600 indeed worked. So that was the unofficial board not supporting this baud rate indeed. Thanks.

Now, concerning the example code, I don't understand how the file "ArduCAM_Mini_2MP_Plus_functions" should work. Precisely how to select the choices in the switch statement. When using the arduino serial console, like I said before, whatever the input I enter in the console, the result is always the same : 
```
ACK CMD CAM start single shoot. END
ACK CMD CAM Capture Done. END
```

UCTRONICS, 2019-11-08:
@Bokoblin 
In fact, this demo will receive different commands to run different functions.
You can get the command from the below code 
f (Serial.available())
{
  temp = Serial.read();
  switch (temp)
  {
    case 0:
      myCAM.OV2640_set_JPEG_size(OV2640_160x120);delay(1000);
      Serial.println(F("ACK CMD switch to OV2640_160x120 END"));
     temp = 0xff;
    break;
    case 1:
      myCAM.OV2640_set_JPEG_size(OV2640_176x144);delay(1000);
      Serial.println(F("ACK CMD switch to OV2640_176x144 END"));
    temp = 0xff;
    break;
    case 2: 
      myCAM.OV2640_set_JPEG_size(OV2640_320x240);delay(1000);
      Serial.println(F("ACK CMD switch to OV2640_320x240 END"));
    temp = 0xff;
    break;
    case 3:
    myCAM.OV2640_set_JPEG_size(OV2640_352x288);delay(1000);
    Serial.println(F("ACK CMD switch to OV2640_352x288 END"));
   temp = 0xff;
    break;
    case 4:
      myCAM.OV2640_set_JPEG_size(OV2640_640x480);delay(1000);
      Serial.println(F("ACK CMD switch to OV2640_640x480 END"));
    temp = 0xff;
    break;
    case 5:
    myCAM.OV2640_set_JPEG_size(OV2640_800x600);delay(1000);
    Serial.println(F("ACK CMD switch to OV2640_800x600 END"));
    temp = 0xff;
    break;
    case 6:
     myCAM.OV2640_set_JPEG_size(OV2640_1024x768);delay(1000);
     Serial.println(F("ACK CMD switch to OV2640_1024x768 END"));
    temp = 0xff;
    break;
    case 7:
    myCAM.OV2640_set_JPEG_size(OV2640_1280x1024);delay(1000);
    Serial.println(F("ACK CMD switch to OV2640_1280x1024 END"));
    temp = 0xff;
    break;
    case 8:
    myCAM.OV2640_set_JPEG_size(OV2640_1600x1200);delay(1000);
    Serial.println(F("ACK CMD switch to OV2640_1600x1200 END"));
     temp = 0xff;
    break;
    case 0x10:
    mode = 1;
    temp = 0xff;
    start_capture = 1;
    Serial.println(F("ACK CMD CAM start single shoot. END"));
    break;
    case 0x11: 
    temp = 0xff;
    myCAM.set_format(JPEG);
    myCAM.InitCAM();
    #if !(defined (OV2640_MINI_2MP_PLUS))
    myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
    #endif
    break;
    case 0x20:
    mode = 2;
    temp = 0xff;
[... 80 lines cut ...]
    myCAM.OV2640_set_Special_effects(Antique);temp = 0xff;
    Serial.println(F("ACK CMD Set to Antique END"));break;
   case 0x81:
    myCAM.OV2640_set_Special_effects(Bluish);temp = 0xff;
    Serial.println(F("ACK CMD Set to Bluish END"));break;
   case 0x82:
    myCAM.OV2640_set_Special_effects(Greenish);temp = 0xff;
    Serial.println(F("ACK CMD Set to Greenish END"));break;  
   case 0x83:
    myCAM.OV2640_set_Special_effects(Reddish);temp = 0xff;
    Serial.println(F("ACK CMD Set to Reddish END"));break;  
   case 0x84:
    myCAM.OV2640_set_Special_effects(BW);temp = 0xff;
    Serial.println(F("ACK CMD Set to BW END"));break; 
  case 0x85:
    myCAM.OV2640_set_Special_effects(Negative);temp = 0xff;
    Serial.println(F("ACK CMD Set to Negative END"));break; 
  case 0x86:
    myCAM.OV2640_set_Special_effects(BWnegative);temp = 0xff;
    Serial.println(F("ACK CMD Set to BWnegative END"));break;   
   case 0x87:
    myCAM.OV2640_set_Special_effects(Normal);temp = 0xff;
    Serial.println(F("ACK CMD Set to Normal END"));break;     
  }
}

Bokoblin, 2019-12-07:
@UCTRONICS Hi, after several weeks of trying to understand why the example **"ArduCAM_Mini_2MP_Plus_Multi_Capture2SD"**  does not work as expected and tweaking it, I have been no clue so far.

More exactly, after setting the proper SD CS pin (mine is 53 on the Mega 2560), everything seems to work until the line 198 :
``
outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
 ``
There, after launching the program, we get _File open failed_ meaning the above line didn't work. 

Do you know how to solve this problem?

PS: For information, the videostreaming example is perfectly working with the Host_V2 application.
So it seems like the camera is Okay but the Camera2SD example has a problem writing to SD card.

---

By the way, I have tweaked the example to a point where the program manages to open the file for writing, however, another problem occurred: the JPG file is completely empty (0 bytes), like if the fifo data wasn't written and flushed to the file.
I have added the example as an attached file if you want to check.
[ArduCAM_Mini_2MP_Plus_Multi_Capture2SD_tweaked.zip](https://github.com/ArduCAM/Arduino/files/3935275/ArduCAM_Mini_2MP_Plus_Multi_Capture2SD_tweaked.zip)

Hardware Info
----------------
Arduino : Arduino Mega 2560 (official)
SD card :  16GB, formatted in FAT32. empty.

Bokoblin, 2019-12-12:
@UCTRONICS  @ArduCAM @ArducamTeam Any update to my comment, please ?