ArduCAM/Arduino #448: ArduCAM::OV5642_set_Mirror_Flip() always result in black picture

sohymg, 2019-10-05:
I am using an Arduino Nano and I noticed that calling `ArduCAM::OV5642_set_Mirror_Flip()` with any value (MIRROR/FLIP/MIRROR_FLIP/Normal) will always result in a black picture. 

```
void setup() {
  ...
  myCAM.set_format(JPEG);
  myCAM.InitCAM();

  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_320x240);
  delay(1000);
  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);

  // does not work, always return black picture
  OV5642_set_Mirror_Flip(FLIP);

  // this works
  uint8_t reg_val;
  myCAM.rdSensorReg16_8(0x3818,&reg_val);
  Serial.println(reg_val); // prints 168
  reg_val = reg_val&0xDF;
  myCAM.wrSensorReg16_8(0x3818 ,reg_val);
}
```

ArducamTeam, 2019-10-06:
@sohymg 
Maybe there is something wrong with the register in the library. We are on the National Day holiday now and there is no equipment test. We will help you to test it as soon as we get back to the company.

ArducamTeam, 2019-10-08:
@sohymg 
Hi,
We have checked the problem you mentioned and used the original configuration. There is indeed a problem with the black picture. The vertical flip function can be used normally, while the mirror function is not so normal.We are looking for a solution.You can do a vertical flip using the method you mentioned above.

patrickmoffitt, 2020-06-05:
I also have this issue. Is there any possibility of a fix?