ArduCAM/Arduino #449: Arducam data processing

Antosuri, 2019-10-05:
I use an Arducam mini 2MP connected to an Arduino to know the position of a black cross line on a white background with an accuracy of tens of microns.
To do this, I need to access a RAW image, with the maximum possible resolution, to locate the center point of the cross.
Is it possible to work with the Arducam mini 2MP in RAW mode with a resolution of 1600 x 1200 pixels?
Is it possible to access the FIFO row by row and column by column through the Arduino ?.
Can someone tell me a script for Arduino to help me?
Thank you.

UCTRONICS, 2019-10-06:
@Antosuri 
At present, we don't have ov2640 RAW config file. we just have RGB and JPEG config files. Why not use the RGB mode? 
You can read the fifo byte by byte.  Notice, if one frame has not be read finished, don't clear the fifo.

Antosuri, 2019-10-06:
It's true, I don't need to use the RAW format, and I can use the RGB format.
To locate the position of the vertical line of the cross, we add all the bytes of each column, and the column that has the highest (or lowest) value corresponds to the column that has the line.
We do the same with the rows, to find the row that contains the horizontal black line.
The pair (column, row) corresponds to the center of the cross.
How can I identify the rows and columns in FIFO?

UCTRONICS, 2019-10-07:
@Antosuri 
Hi, It does not have the signal of each column or rows. The fifo will stored one frame and you can read it byte by byte.

Antosuri, 2019-10-08:
Thanks for the reply.
Okay, I will use the RGB format to read the image, and I will read the FIFO byte by byte, but I need the maximum resolution of 1600x1200, and I see that this resolution is available for the JPEG format, but I can't find it for the BMP format.
How should I program the camera to get the resolution of 1600x1200 in BMP?

Splitframe, 2019-10-08:
We use the 5MP version and also would like to access RAW values for specific regions at full resolution.
It would be very helpful if there was an option to directly read from the sensor, bypassing the FIFO memory. Even if this means rolling shutter due to a slower scan rate.

UCTRONICS, 2019-10-09:
@Splitframe 
We don't have the 1600x1200 resolution configuration. You can refer to this document 
[OV2640 Camera Module Software Application Notes1.04 OVT.pdf](https://github.com/ArduCAM/Arduino/files/3705010/OV2640.Camera.Module.Software.Application.Notes1.04.OVT.pdf)

Splitframe, 2019-10-09:
@UCTRONICS I think you meant @Antosuri

UCTRONICS, 2019-10-09:
@Splitframe 
It does not support bypassing the FIFO. On Arduino platform, we just read the image data through the SPI, so the image data must come into the FIFO.

Antosuri, 2019-10-09:
We are working on an Open-Source project to monitor the cracks of churches and castles, and we need an accuracy of the order of tens of microns.
For this we take a high resolution photograph of the crack (minimum 1600x1200 pixels), and by means of a simple image processing routine we can know the distance that the crack has been separated, and monitor its movement in real time.
The image process is carried out by means of an arduino pro mini with the objective of minimizing consumption since the equipment must be small and autonomous.
For this we need to process a BMP file of the photograph. We don't use a JPEG file, since we don't have the possibility to convert it to BMP.
We have seen that ArduCAM does not have drivers for the resolution of 1600x1200 in BMP format and, therefore, does not work for the project.
Do you know other cameras with a minimum resolution of 1600x1200 that have the drivers for the BMP format, and that can be easily connected to an Arduino or an ESP?

UCTRONICS, 2019-10-10:
@Antosuri @Splitframe 
Hi, Our Arducam mini 5mp plus support getting 2592x1944 raw format image. Is it help for your project?

Antosuri, 2019-10-10:
It can work if we can read the entire RAW image.
We have bought a unit of the 5MP Plus model, and we will test it when it arrives.
When we have reliable results we will update the thread.
Thank you