ArduCAM/Arduino #452: ArduCAM 2MP mini Plus QVGA bitmap color issue

xiongyu0523, 2019-11-02:
I try to save the QVGA raw data into a 16-bit bitmap file. I'm using the default QVGA setting from Arduino project. The image I get dump out have a lot of 'blue' noise points but captured color seems correct, so I think figured out how to save raw data into a 16-bit color bitmap file correctly.

Anybody knows what is the potential reason lead to this?

Please check 3 snapshots I dumped out, I convert to PNG here because Github does not support bitmap file preview, the effects are the same. 

(BTW, My camera driver works perfect under JPEG mode)

[image]
[image]
[image]

xiongyu0523, 2019-11-02:
Here is my code to save into bitmap mode. I search a lot on internet and find a reference file to learn from. 

p_imgBuffer store the 320x240x2 bytes raw data read from ArduCAM camera.

```
	const uint8_t cBitmapFileHeaderSize = 14;
	const uint8_t cBitmapInfoHeaderSize = 40;
	const uint8_t cBitmapbmiColorsSize = 12;

	uint32_t width  = 320;
	uint32_t height = 240;
	uint32_t byteperpixel = 2;

	file_size = cBitmapFileHeaderSize + cBitmapInfoHeaderSize + cBitmapbmiColorsSize + img_len;
	p_file = calloc(file_size, 1);

	// mark 'BM'
	p_file[0]  = 'B';
	p_file[1]  = 'M';

	// total file size
	p_file[2]  = (uint8_t)(file_size);
	p_file[3]  = (uint8_t)(file_size >> 8);
	p_file[4]  = (uint8_t)(file_size >> 16);
	p_file[5]  = (uint8_t)(file_size >> 24);

	// reserved
	p_file[6] = 0x00;
	p_file[7] = 0x00;
	p_file[8] = 0x00;
	p_file[9] = 0x00;

	// Pixel Array offset
	p_file[10] = (uint8_t)(cBitmapFileHeaderSize + cBitmapInfoHeaderSize + cBitmapbmiColorsSize);
	p_file[11] = 0x00;
	p_file[12] = 0x00;
	p_file[13] = 0x00;

	p_file[14] = cBitmapInfoHeaderSize;
	p_file[15] = 0x00;
	p_file[16] = 0x00;
	p_file[17] = 0x00;

	p_file[18] = (uint8_t)(width);
	p_file[19] = (uint8_t)(width >> 8);
	p_file[20] = (uint8_t)(width >> 16);
	p_file[21] = (uint8_t)(width >> 24);

	p_file[22] = (uint8_t)(height);
	p_file[23] = (uint8_t)(height >> 8);
	p_file[24] = (uint8_t)(height >> 16);
	p_file[25] = (uint8_t)(height >> 24);

	// biPlanes = 0x01
	p_file[26] = 0x01;
	p_file[27] = 0x00;

	// biBitCount = 16
	p_file[28] = byteperpixel * 8;
	p_file[29] = 0x00;

	// biCompression = BI_BITFIELDS
	p_file[30] = 0x03;
	p_file[31] = 0x00;
	p_file[32] = 0x00;
	p_file[33] = 0x00;

	// biSizeImage = 153600 (320 x 240 x 2)
	p_file[34] = 0x00;
	p_file[35] = 0x58;
	p_file[36] = 0x02;
	p_file[37] = 0x00;

	// biXPelsPerMeter = 0
	p_file[38] = 0x00;
	p_file[39] = 0x00;
	p_file[40] = 0x00;
	p_file[41] = 0x00;

	// biYPelsPerMeter = 0
	p_file[42] = 0x00;
	p_file[43] = 0x00;
	p_file[44] = 0x00;
	p_file[45] = 0x00;

	// biClrUsed = 0
	p_file[46] = 0x00;
	p_file[47] = 0x00;
	p_file[48] = 0x00;
	p_file[49] = 0x00;

	// biClrImportant = 0;
	p_file[50] = 0x00;
	p_file[51] = 0x00;
	p_file[52] = 0x00;
	p_file[53] = 0x00;

	// data is organized as little end RGB565, MSB first
	// RedMask
	p_file[54] = 0xF8;
	p_file[55] = 0x00;
	p_file[56] = 0x00;
	p_file[57] = 0x00;

	// Green Mask
	p_file[58] = 0x07;
	p_file[59] = 0xE0;
	p_file[60] = 0x00;
	p_file[61] = 0x00;

	// Blue Mask
	p_file[62] = 0x00;
	p_file[63] = 0x1F;
	p_file[64] = 0x00;
	p_file[65] = 0x00;

	// assume no padding if required for simplicity
	memcpy(&p_file[cBitmapFileHeaderSize + cBitmapInfoHeaderSize + cBitmapbmiColorsSize], p_imgBuffer, img_len);
	
	free(p_imgBuffer);
```

ArducamTeam, 2019-11-05:
Hi, it may be your bmp header is not correct, you can refer to the bmp header in our example, the link is: https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Camera_Playback

xiongyu0523, 2019-11-06:
Thank you.

I found the problem is caused by swapping order of pixel bytes. When reading from ArduCAM memory, we get MSB first and LSB later for one 16-bit pixel. I'm aware of this and try to change the Red/Green/Blue mask order, and keep the pixel order. But this ways seems not OK. We have to organize the pixel data in LSB MSB LSB MSB pattern. 

One problem I found in the official example is the file size is incorrect. It should be 153666, not 153654.

xiongyu0523, 2019-11-06:
We can close this ticket.

closed 2019-11-06 by xiongyu0523