ArduCAM/Arduino #454: 'regtype' does not name a type error

canerpehlivan, 2019-11-04:
Hello,

I am using an STM32L073RZ board and The Avnet Silica NB-IoT Sensor Shield(BG96 Queltec) with Arduino IDE and I want to connect the OV5642 camera module to it. I downloaded necessary libraries and Arduino IDE is correctly working with the board(I can blink an LED). When it comes to the ArduCAM this is the error:

> Arduino: 1.8.10 (Windows 10), Board: "Nucleo-64, Nucleo L073RZ, Mass Storage, Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"
> 
> A subdirectory or file sketch already exists.
> 
> In file included from C:\Users\caner\AppData\Local\Temp\arduino_modified_sketch_7921\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino:1:
> C:\Program Files (x86)\Arduino\libraries\ArduCAM/ArduCAM.h:752:2: error: 'regtype' does not name a type; did you mean 'decltype'?
>   regtype *P_CS;
>   ^~~~~~~
>   decltype
> C:\Program Files (x86)\Arduino\libraries\ArduCAM/ArduCAM.h:753:2: error: 'regsize' does not name a type
>   regsize B_CS;
>   ^~~~~~~
> Multiple libraries were found for "ArduCAM.h"
>  Used: C:\Program
> Multiple libraries were found for "Wire.h"
>  Used: C:\Users\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\Wire
> Multiple libraries were found for "SPI.h"
>  Used: C:\Users\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SPI
> exit status 1
> Error compiling for board Nucleo-64.
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

I am not sure what is the problem but it looks like ArduCAM modules are not supported in STM32 with Arduino IDE.

UCTRONICS, 2019-11-05:
@canerpehlivan 
Yes, ArduCAM is not compatible with STM32 chip on Arduino IDE.

canerpehlivan, 2019-11-05:
Thank you for the fast response. I want to use uKeil Vision then, but the example shown in STM32 repo is for another board type. I need to convert it to my board, but I need .ico file for Cube MX which includes configurations for timers, SPI, I2C and others. How can I get it?

UCTRONICS, 2019-11-06:
@canerpehlivan 
I don't have cube MX project. Why not use our stm32 project directly? 
If you want to use cube MX, you need to refer to our SPI and I2C driver to build your cube project.

canerpehlivan, 2019-11-06:
@UCTRONICS 
Thank you for the reply.
Because my board part number is different than your stm32 project(stm32l073rz), and I don’t really know how to convert your project according to my board in Keil uVision 5. Is there a way to compile it in my board?

UCTRONICS, 2019-11-06:
@canerpehlivan 
My project is build on STM32F103C8T6, You are right, it does need fix the driver to support for your board. I don't have the stm32l073rz board. But I can help you fix it step by step. 
Firstly, You should change the i2c bus. In fact, I use the software i2c, you just need to use the gpio to make it.

canerpehlivan, 2019-11-06:
My board also supports the i2c, why I should use gpio? Also, If I am going to implement everything on Keil, how can I adjust pin configurations without using STM32CubeMX software? Is it possible in Keil?

UCTRONICS, 2019-11-07:
@canerpehlivan 
Yes of course it is possible in keil. You need to write the i2c and spi driver?
In our code I useb GPIO to simulate i2 bus which is easy to migrate to your platform.

canerpehlivan, 2019-11-08:
@UCTRONICS thank you for the answer. Again I am not sure how to migrate all these to another platform. Maybe changing board type in Keil is a better option? When I change board type I get a lot of errors. How can I delete those stm32f0x files from your project and make them stm32l0?