ArduCAM/Arduino #455: Arducam do not work change resolution

Mark-Jang, 2019-11-06:
Hello.

We're creating an app that transmits photos to Android applications through Bluetooth.

Previously, photo transmission was facilitated with a resolution of 1024x768.

After increasing the resolution to 1920x1080, no matter what operation, only 320x240. I only increased the resolution of a previously used source.

It is not being transmitted at 1024x768.

How do I solve these problems?

Mark-Jang, 2019-11-06:
I use Arducam Mini OV5642.

UCTRONICS, 2019-11-07:
@Mark-Jang 
Which demo are you using? What's your hardware platform?

Mark-Jang, 2019-11-07:
My hardware is  Arduino Nano.

ArducamTeam, 2019-11-07:
@Mark-Jang 
Hi, can you paste in the code you used?We're not sure if the problem is your code or our library.

Mark-Jang, 2019-11-07:
This is My code

// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can test OV5642 functions.
//

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

#include <SoftwareSerial.h>
SoftwareSerial BTSerial(3, 4); // RX | TX

#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};

// constants won't change. They're used here to set pin numbers:
const int buttonPin[] = {9, 8, 7, 6, 5};
int buttonState = 0;

// set pin 7 as the slave select for the digital pot:
const int CS = 10;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
ArduCAM myCAM(OV5642, CS);
uint8_t read_fifo_burst(ArduCAM myCAM);

void setup()
{
  uint8_t vid, pid;
  uint8_t temp;

  Wire.begin();

  // Set baudrate - arduino nano
  Serial.begin(230400);
  BTSerial.begin(230400);delay(500);

  for(int i =0; i<5; i++){
    pinMode(buttonPin[i], INPUT_PULLUP );
  }

  BTSerial.print("ACK CMD ArduCAM Start! END\n");
  // set the CS as an output:
  pinMode(CS, OUTPUT);
  digitalWrite(CS, HIGH);
  // initialize SPI:
  SPI.begin();
  //Reset the CPLD
  myCAM.write_reg(0x07, 0x80);
  delay(100);
  myCAM.write_reg(0x07, 0x00);
  delay(100);
  while (1) {
    //Check if the ArduCAM SPI bus is OK
    myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
    temp = myCAM.read_reg(ARDUCHIP_TEST1);
    if (temp != 0x55) {
      BTSerial.print("ACK CMD SPI interface Error! END\n");
      delay(1000); continue;
    }
    else {
      BTSerial.print("ACK CMD SPI interface OK. END\n"); break;
    }
  }
  while (1) {
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    if ((vid != 0x56) || (pid != 0x42)) {
      BTSerial.print("ACK CMD Can't find OV5642 module! END\n");
      delay(1000); continue;
    }
    else {
      BTSerial.print("ACK CMD OV5642 detected. END\n"); break;
    }
  }
  //Change to JPEG capture mode and initialize the OV5642 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();

  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_1600x1200);
  delay(1000);
  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);

}

void loop()
{
  
  if (mode == 0) {
    for(int x = 0; x <5; x++){
      buttonState = digitalRead(buttonPin[x]);
      
     if(buttonState == LOW && buttonPin[x] == 6) {
        BTSerial.print("LeftBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 7) {
        BTSerial.print("RightBtn\n");
        delay(500);
     } 
     if(buttonState == LOW && buttonPin[x] == 8) {
        BTSerial.print("OkBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 9 ) {
        BTSerial.print("HomeBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 5) {
        BTSerial.print("CameraBtn\n");
        mode = 1;
     }
    }
  }
  else if (mode == 1) {
      //BTSerial.begin(115200); delay(2000);
      start_capture = 1;
      mode = 2;
  }
  else if (mode == 2) {
    
    if (start_capture == 1){
      myCAM.flush_fifo();
      myCAM.clear_fifo_flag();
      //Start capture
      myCAM.start_capture();
      start_capture = 0;
    }
    
    if (myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK))
    {
      //BTSerial.print(F("ACK CMD CAM Capture Done. END\n")); delay(50);
      read_fifo_burst(myCAM);
      //Clear the capture done flag
      myCAM.clear_fifo_flag();

      delay(50);
      //BTSerial.begin(9600); delay(2000);
      mode = 0;
    }
  }
}

uint8_t read_fifo_burst(ArduCAM myCAM)
{
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  length = myCAM.read_fifo_length();
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //512 kb
  {
    //BTSerial.print(F("ACK CMD Over size. END\n"));
    return 0;
  }
  if (length == 0) //0 kb
  {
    //BTSerial.print(F("ACK CMD Size is 0. END\n"));
    return 0;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  temp = SPI.transfer(0x00);
  length--;
  while (length--)
  {
    temp_last = temp;
    temp = SPI.transfer(0x00);
    if (is_header == true)
    {
      BTSerial.write(temp);
      //Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    { 
      is_header = true;
      //BTSerial.print(F("ACK IMG END\n"));
      BTSerial.write(temp_last);
      //Serial.write(temp_last);
      BTSerial.write(temp);
      //Serial.write(temp);
    }
    if ((temp == 0xD9) && (temp_last == 0xFF)) //If find the end ,break while,
      break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}

Mark-Jang, 2019-11-07:
`// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can test OV5642 functions.
//

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

#include <SoftwareSerial.h>
SoftwareSerial BTSerial(3, 4); // RX | TX

#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};

// constants won't change. They're used here to set pin numbers:
const int buttonPin[] = {9, 8, 7, 6, 5};
int buttonState = 0;

// set pin 7 as the slave select for the digital pot:
const int CS = 10;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
ArduCAM myCAM(OV5642, CS);
uint8_t read_fifo_burst(ArduCAM myCAM);

void setup()
{
  uint8_t vid, pid;
  uint8_t temp;

  Wire.begin();

  // Set baudrate - arduino nano
  Serial.begin(230400);
  BTSerial.begin(230400);delay(500);

  for(int i =0; i<5; i++){
    pinMode(buttonPin[i], INPUT_PULLUP );
  }

  BTSerial.print("ACK CMD ArduCAM Start! END\n");
  // set the CS as an output:
  pinMode(CS, OUTPUT);
  digitalWrite(CS, HIGH);
  // initialize SPI:
  SPI.begin();
  //Reset the CPLD
  myCAM.write_reg(0x07, 0x80);
  delay(100);
  myCAM.write_reg(0x07, 0x00);
  delay(100);
  while (1) {
    //Check if the ArduCAM SPI bus is OK
    myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
    temp = myCAM.read_reg(ARDUCHIP_TEST1);
    if (temp != 0x55) {
      BTSerial.print("ACK CMD SPI interface Error! END\n");
      delay(1000); continue;
    }
    else {
      BTSerial.print("ACK CMD SPI interface OK. END\n"); break;
    }
  }
  while (1) {
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    if ((vid != 0x56) || (pid != 0x42)) {
      BTSerial.print("ACK CMD Can't find OV5642 module! END\n");
      delay(1000); continue;
    }
    else {
      BTSerial.print("ACK CMD OV5642 detected. END\n"); break;
    }
  }
  //Change to JPEG capture mode and initialize the OV5642 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();

  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_1600x1200);
  delay(1000);
  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);

}

void loop()
{
  
  if (mode == 0) {
    for(int x = 0; x <5; x++){
      buttonState = digitalRead(buttonPin[x]);
      
     if(buttonState == LOW && buttonPin[x] == 6) {
        BTSerial.print("LeftBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 7) {
        BTSerial.print("RightBtn\n");
        delay(500);
     } 
     if(buttonState == LOW && buttonPin[x] == 8) {
        BTSerial.print("OkBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 9 ) {
        BTSerial.print("HomeBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 5) {
        myCAM.set_format(JPEG);
        myCAM.InitCAM();
        myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
        myCAM.OV5642_set_JPEG_size(OV5642_1600x1200);delay(1000);
        BTSerial.print("CameraBtn\n");
        mode = 1;
     }
    }
  }
  else if (mode == 1) {
      //BTSerial.begin(115200); delay(2000);
      start_capture = 1;
      mode = 2;
  }
  else if (mode == 2) {
    
    if (start_capture == 1){
      myCAM.flush_fifo();
      myCAM.clear_fifo_flag();
      //Start capture
      myCAM.start_capture();
      start_capture = 0;
    }
    
    if (myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK))
    {
      //BTSerial.print(F("ACK CMD CAM Capture Done. END\n")); delay(50);
      read_fifo_burst(myCAM);
      //Clear the capture done flag
      myCAM.clear_fifo_flag();

      delay(50);
      //BTSerial.begin(9600); delay(2000);
      mode = 0;
    }
  }
}

uint8_t read_fifo_burst(ArduCAM myCAM)
{
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  length = myCAM.read_fifo_length();
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //512 kb
  {
    //BTSerial.print(F("ACK CMD Over size. END\n"));
    return 0;
  }
  if (length == 0) //0 kb
  {
    //BTSerial.print(F("ACK CMD Size is 0. END\n"));
    return 0;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  temp = SPI.transfer(0x00);
  length--;
  while (length--)
  {
    temp_last = temp;
    temp = SPI.transfer(0x00);
    if (is_header == true)
    {
      BTSerial.write(temp);
      //Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    { 
      is_header = true;
      //BTSerial.print(F("ACK IMG END\n"));
      BTSerial.write(temp_last);
      //Serial.write(temp_last);
      BTSerial.write(temp);
      //Serial.write(temp);
    }
    if ((temp == 0xD9) && (temp_last == 0xFF)) //If find the end ,break while,
      break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}`

Mark-Jang, 2019-11-07:
This is my code.

// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can test OV5642 functions.
//

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

#include <SoftwareSerial.h>
SoftwareSerial BTSerial(3, 4); // RX | TX

#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};

// constants won't change. They're used here to set pin numbers:
const int buttonPin[] = {9, 8, 7, 6, 5};
int buttonState = 0;

// set pin 7 as the slave select for the digital pot:
const int CS = 10;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
ArduCAM myCAM(OV5642, CS);
uint8_t read_fifo_burst(ArduCAM myCAM);

void setup()
{
  uint8_t vid, pid;
  uint8_t temp;

  Wire.begin();

  // Set baudrate - arduino nano
  Serial.begin(230400);
  BTSerial.begin(230400);delay(500);

  for(int i =0; i<5; i++){
    pinMode(buttonPin[i], INPUT_PULLUP );
  }

  BTSerial.print("ACK CMD ArduCAM Start! END\n");
  // set the CS as an output:
  pinMode(CS, OUTPUT);
  digitalWrite(CS, HIGH);
  // initialize SPI:
  SPI.begin();
  //Reset the CPLD
  myCAM.write_reg(0x07, 0x80);
  delay(100);
  myCAM.write_reg(0x07, 0x00);
  delay(100);
  while (1) {
    //Check if the ArduCAM SPI bus is OK
    myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
    temp = myCAM.read_reg(ARDUCHIP_TEST1);
    if (temp != 0x55) {
      BTSerial.print("ACK CMD SPI interface Error! END\n");
      delay(1000); continue;
    }
    else {
      BTSerial.print("ACK CMD SPI interface OK. END\n"); break;
    }
  }
  while (1) {
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    if ((vid != 0x56) || (pid != 0x42)) {
      BTSerial.print("ACK CMD Can't find OV5642 module! END\n");
      delay(1000); continue;
    }
    else {
      BTSerial.print("ACK CMD OV5642 detected. END\n"); break;
    }
  }
  //Change to JPEG capture mode and initialize the OV5642 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();

  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_1600x1200);
  delay(1000);
  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);

}

void loop()
{
  
  if (mode == 0) {
    for(int x = 0; x <5; x++){
      buttonState = digitalRead(buttonPin[x]);
      
     if(buttonState == LOW && buttonPin[x] == 6) {
        BTSerial.print("LeftBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 7) {
        BTSerial.print("RightBtn\n");
        delay(500);
     } 
     if(buttonState == LOW && buttonPin[x] == 8) {
        BTSerial.print("OkBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 9 ) {
        BTSerial.print("HomeBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 5) {
        myCAM.set_format(JPEG);
        myCAM.InitCAM();
        myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
        myCAM.OV5642_set_JPEG_size(OV5642_1600x1200);delay(1000);
        BTSerial.print("CameraBtn\n");
        mode = 1;
     }
    }
  }
  else if (mode == 1) {
      //BTSerial.begin(115200); delay(2000);
      start_capture = 1;
      mode = 2;
  }
  else if (mode == 2) {
    
    if (start_capture == 1){
      myCAM.flush_fifo();
      myCAM.clear_fifo_flag();
      //Start capture
      myCAM.start_capture();
      start_capture = 0;
    }
    
    if (myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK))
    {
      //BTSerial.print(F("ACK CMD CAM Capture Done. END\n")); delay(50);
      read_fifo_burst(myCAM);
      //Clear the capture done flag
      myCAM.clear_fifo_flag();

      delay(50);
      //BTSerial.begin(9600); delay(2000);
      mode = 0;
    }
  }
}

uint8_t read_fifo_burst(ArduCAM myCAM)
{
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  length = myCAM.read_fifo_length();
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //512 kb
  {
    //BTSerial.print(F("ACK CMD Over size. END\n"));
    return 0;
  }
  if (length == 0) //0 kb
  {
    //BTSerial.print(F("ACK CMD Size is 0. END\n"));
    return 0;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  temp = SPI.transfer(0x00);
  length--;
  while (length--)
  {
    temp_last = temp;
    temp = SPI.transfer(0x00);
    if (is_header == true)
    {
      BTSerial.write(temp);
      //Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    { 
      is_header = true;
      //BTSerial.print(F("ACK IMG END\n"));
      BTSerial.write(temp_last);
      //Serial.write(temp_last);
      BTSerial.write(temp);
      //Serial.write(temp);
    }
    if ((temp == 0xD9) && (temp_last == 0xFF)) //If find the end ,break while,
      break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}

Mark-Jang, 2019-11-07:
This is my code

```
// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can test OV5642 functions.
//

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

#include <SoftwareSerial.h>
SoftwareSerial BTSerial(3, 4); // RX | TX

#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};

// constants won't change. They're used here to set pin numbers:
const int buttonPin[] = {9, 8, 7, 6, 5};
int buttonState = 0;

// set pin 7 as the slave select for the digital pot:
const int CS = 10;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
ArduCAM myCAM(OV5642, CS);
uint8_t read_fifo_burst(ArduCAM myCAM);

void setup()
{
  uint8_t vid, pid;
  uint8_t temp;

  Wire.begin();

  // Set baudrate - arduino nano
  Serial.begin(230400);
  BTSerial.begin(230400);delay(500);

  for(int i =0; i<5; i++){
    pinMode(buttonPin[i], INPUT_PULLUP );
  }

  BTSerial.print("ACK CMD ArduCAM Start! END\n");
  // set the CS as an output:
  pinMode(CS, OUTPUT);
  digitalWrite(CS, HIGH);
  // initialize SPI:
  SPI.begin();
  //Reset the CPLD
  myCAM.write_reg(0x07, 0x80);
  delay(100);
  myCAM.write_reg(0x07, 0x00);
  delay(100);
  while (1) {
    //Check if the ArduCAM SPI bus is OK
    myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
    temp = myCAM.read_reg(ARDUCHIP_TEST1);
    if (temp != 0x55) {
      BTSerial.print("ACK CMD SPI interface Error! END\n");
      delay(1000); continue;
    }
    else {
      BTSerial.print("ACK CMD SPI interface OK. END\n"); break;
    }
  }
  while (1) {
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    if ((vid != 0x56) || (pid != 0x42)) {
      BTSerial.print("ACK CMD Can't find OV5642 module! END\n");
      delay(1000); continue;
    }
    else {
      BTSerial.print("ACK CMD OV5642 detected. END\n"); break;
    }
  }
  //Change to JPEG capture mode and initialize the OV5642 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();

  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_1600x1200);
  delay(1000);
  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);

}

void loop()
{
  
  if (mode == 0) {
    for(int x = 0; x <5; x++){
      buttonState = digitalRead(buttonPin[x]);
      
     if(buttonState == LOW && buttonPin[x] == 6) {
        BTSerial.print("LeftBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 7) {
        BTSerial.print("RightBtn\n");
        delay(500);
     } 
     if(buttonState == LOW && buttonPin[x] == 8) {
        BTSerial.print("OkBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 9 ) {
        BTSerial.print("HomeBtn\n");
        delay(500);
     }
     if(buttonState == LOW && buttonPin[x] == 5) {
        myCAM.set_format(JPEG);
        myCAM.InitCAM();
        myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
        myCAM.OV5642_set_JPEG_size(OV5642_1600x1200);delay(1000);
        BTSerial.print("CameraBtn\n");
        mode = 1;
     }
    }
  }
  else if (mode == 1) {
      //BTSerial.begin(115200); delay(2000);
      start_capture = 1;
      mode = 2;
  }
  else if (mode == 2) {
    
    if (start_capture == 1){
      myCAM.flush_fifo();
      myCAM.clear_fifo_flag();
      //Start capture
      myCAM.start_capture();
      start_capture = 0;
    }
    
    if (myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK))
    {
      //BTSerial.print(F("ACK CMD CAM Capture Done. END\n")); delay(50);
      read_fifo_burst(myCAM);
      //Clear the capture done flag
      myCAM.clear_fifo_flag();

      delay(50);
      //BTSerial.begin(9600); delay(2000);
      mode = 0;
    }
  }
}

uint8_t read_fifo_burst(ArduCAM myCAM)
{
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  length = myCAM.read_fifo_length();
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //512 kb
  {
    //BTSerial.print(F("ACK CMD Over size. END\n"));
    return 0;
  }
  if (length == 0) //0 kb
  {
    //BTSerial.print(F("ACK CMD Size is 0. END\n"));
    return 0;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  temp = SPI.transfer(0x00);
  length--;
  while (length--)
  {
    temp_last = temp;
    temp = SPI.transfer(0x00);
    if (is_header == true)
    {
      BTSerial.write(temp);
      //Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    { 
      is_header = true;
      //BTSerial.print(F("ACK IMG END\n"));
      BTSerial.write(temp_last);
      //Serial.write(temp_last);
      BTSerial.write(temp);
      //Serial.write(temp);
    }
    if ((temp == 0xD9) && (temp_last == 0xFF)) //If find the end ,break while,
      break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}
```

ArducamTeam, 2019-11-07:
@Mark-Jang 
You mentioned in the question that changing the resolution didn't work, but I looked at your code and it only used one resolution and didn't change.
If you want to confirm the resolution issue, you can test this demo ‘ArduCAM_Mini_5MP_Plus_Video_Streaming’  with our software.
Demo link:[https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_5MP_Plus_Video_Streaming](url)
Software link:[https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows](url)

Mark-Jang, 2019-11-07:
I was able to adjust the resolution freely in the demo I was given.

What I want is to use only one resolution.

It's transferring photos to Android.

No matter what resolution you set, only 320x240.

ArducamTeam, 2019-11-07:
First I suggest you use our software to make sure the resolution of the Settings is consistent with the output.If there is no problem with the resolution of the output, it is possible that the resolution has been cropped on the android .

Mark-Jang, 2019-11-07:
Thank you.

I re-created my source code from a given code, and I re-installed the Arducam library.

The result is that it worked well.

closed 2019-11-07 by Mark-Jang