ArduCAM/Arduino #457: is there a way to parallel image wait and CPLD read operation?

xiongyu0523, 2019-11-20:
Now I'm using single shot mode. It takes ~100ms to get a 320x240 image ready and ~150ms for me to read it out via 8M SPI. I want to make this two steps in parallel.  When I try to adjust the sequence of operation, I can't figure out how to deal with the clear_fifo_flag() operation. 

Below is a pseudo process.
```
while(1)
    1. Trigger
    2. wait for image ready (100ms)
    3. read length
    4. burst read data (150ms)
    5. clear fifo
```

Change to
```
0. Trigger (first image)
while(1)
    1. wait for image ready (only first time need 100ms)
    2. read length
    3. trigger (a new image)
    4. burst read data (150ms)
    5. clear fifo (this is key..if I do not clear fifo, I will get corrupt data, if I clear fifo I will lost the new image and wait at 1) forever)
```

ArducamTeam, 2019-11-21:
CPLD does not support this synchronization mode currently , and we may develop this mode in a future release.

UCTRONICS, 2019-11-22:
@xiongyu0523 
If the quantity required is large, we can customer it for you.

xiongyu0523, 2019-11-22:
Thank you. I'm just doing a prototype.

closed 2019-11-22 by xiongyu0523