ArduCAM/Arduino #458: Arucam MT9D111 Digital Camera Example Invalid JPEG

joehowarth17, 2019-11-26:
Hi,

I am trying to save an image to the SD card on the ArduCAM Shield, using the ArduCAM_Shield_V2_Digital_Camera.ino sketch.
For hardware, I am using the ArduCAM shield V2 with the MT9D111 camera and an Arduino Due.

The issue is that I end up with multiple zero size images on the SD card, which cannot be opened.
This seems to be happening because the code reading in the image never receives the JPEG SOI(start of image) marker,0xFF 0xD8, and so it never saves the bytes to the SD card, 
resulting in an empty JPEG file on the SD card.

line 298:
```    
else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      buf[i++] = temp_last;
      buf[i++] = temp;   
    } 
```

I am running the example program exactly, I only modified the memorysaver.h selecting ARDUCAM_SHIELD_V2, and MT9D111A_CAM.

What I have tried:
	- I un-commented the code that checks for the JPEG SOI marker, and looked at the image bytes. There is no SOI marker or any of the other JPEG markers. I am not convinced that the image is in JPEG format.
	- I have tried lowering the SPI speed, there is no difference except it takes longer to transfer the bytes.
	- I have tried with 2 different ArduCAM Shield V2, 2 different MT9D111 cameras and 2 different Arduino Dues. No combination results in proper operation. 

Does anyone know how to solve this issue?
Does the MT9D111 work with this sketch and is there any other example code that will work with this camera? Many of the examples don't seem to support this camera.

BillyBee123, 2020-02-22:
The first line should be a double ampersand (&&) and not a single ampersand (&). This is, unfortuantly, a recurring bug in the library examples, and has been noted by others (https://github.com/ArduCAM/Arduino/issues/429).

Double ampersand is a logical AND operator that evaluates two argument conditions.  Double ampersand returns a logical value of either TRUE or FALSE.  For example, (3==3 && 4==3) = FALSE.

Single ampersand is a bitwise AND operator that compares two argument values expressed in binary.  It applies the AND operator successively to the argument bits, and returns a numeric value.  For example, 0011 & 0101 = 0001.

The code above will compile if either a single ampersand or double ampersand is used.  However, the single ampersand will cause unexpected and incorrect results.  This will hopefully resolve your issue.

ArducamTeam, 2020-02-27:
@joehowarth17 
Hi,
We wrote a MT9D111 camera example in ARDUCAM_SHIELD_V2 platform, due to the jpeg images can't directly to the LCD, so to save images to the SD card, you can see the images init. You can adjust the resolution in the example, and adjust the  OFFSET, in order to avoid gray lines in the end of image.The gray line may sometimes appear at the end of the image, and we're trying to optimize it.
`// ArduCAM demo (C)2020 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions of the library with a supported camera modules.
// This demo can only work on ARDUCAM_SHIELD_V2 platform.
// This demo will run the MT9D111 with JPEG mode, it is compatible with autofocus and non-autofocus.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG output mode.
// 2. Capture and buffer the image to FIFO. 
// 3. Store the image to Micro SD/TF card with JPEG format.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM shield V2
// and use Arduino IDE 1.8.9 compiler or above.

#include <UTFT_SPI.h>
#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#if defined(__arm__)
#include <itoa.h>
#endif

#define SD_CS 9
#define JPEGIMAGEOFFSET 625

  //  w1 = 0x0140;  320
  //  h1 = 0x00f0;  240
  //  w2 = 0x0320;  800
  //  h2 = 0x0258;  600
  //  w2 = 0x0640;  1600
  //  h2 = 0x04B0;  1200
  
int jpeg = 1;
unsigned long jpeg_length = 0;
// Set the offset to adjust the error data at the end of the image.
int16_t OFFSET = 50;
//set resolution 
const uint16_t width = 1600;
const uint16_t height = 1200;
// set pin 10 as the slave select for the ArduCAM shiel:
const int slaveSelectPin = 10;

const char JPEG_header[625] PROGMEM =
{
  0xff, 0xd8, 0xff, 0xe0, 0x00, 0x10, 0x4a, 0x46, 0x49, 0x46, 0x00, 0x01, 0x02, 0x00, 0x00, 0x01,
  0x00, 0x01, 0x00, 0x00, 0xff, 0xdb, 0x00, 0x84, 0x00, 0x03, 0x02, 0x02, 0x03, 0x02, 0x02, 0x03,
  0x03, 0x02, 0x03, 0x03, 0x03, 0x03, 0x04, 0x05, 0x08, 0x05, 0x05, 0x04, 0x04, 0x05, 0x09, 0x07,
  0x07, 0x05, 0x08, 0x0b, 0x0a, 0x0b, 0x0b, 0x0b, 0x0a, 0x0b, 0x0a, 0x0c, 0x0e, 0x11, 0x0f, 0x0c,
  0x0d, 0x10, 0x0d, 0x0a, 0x0b, 0x0f, 0x14, 0x0f, 0x10, 0x12, 0x12, 0x13, 0x14, 0x13, 0x0c, 0x0e,
  0x15, 0x17, 0x15, 0x13, 0x17, 0x11, 0x13, 0x13, 0x13, 0x01, 0x03, 0x03, 0x03, 0x05, 0x04, 0x05,
  0x09, 0x05, 0x05, 0x09, 0x13, 0x0c, 0x0b, 0x0c, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13,
  0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13,
  0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13,
  0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0x13, 0xff, 0xc0, 0x00, 0x11, 0x08, 0x04,
  0x80, 0x06, 0x40, 0x03, 0x00, 0x21, 0x00, 0x01, 0x11, 0x01, 0x02, 0x11, 0x01, 0xff, 0xc4, 0x00,
  0x1f, 0x00, 0x00, 0x01, 0x05, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0a, 0x0b, 0xff, 0xc4,
  0x00, 0xb5, 0x10, 0x00, 0x02, 0x01, 0x03, 0x03, 0x02, 0x04, 0x03, 0x05, 0x05, 0x04, 0x04, 0x00,
  0x00, 0x01, 0x7d, 0x01, 0x02, 0x03, 0x00, 0x04, 0x11, 0x05, 0x12, 0x21, 0x31, 0x41, 0x06, 0x13,
  0x51, 0x61, 0x07, 0x22, 0x71, 0x14, 0x32, 0x81, 0x91, 0xa1, 0x08, 0x23, 0x42, 0xb1, 0xc1, 0x15,
  0x52, 0xd1, 0xf0, 0x24, 0x33, 0x62, 0x72, 0x82, 0x09, 0x0a, 0x16, 0x17, 0x18, 0x19, 0x1a, 0x25,
  0x26, 0x27, 0x28, 0x29, 0x2a, 0x34, 0x35, 0x36, 0x37, 0x38, 0x39, 0x3a, 0x43, 0x44, 0x45, 0x46,
  0x47, 0x48, 0x49, 0x4a, 0x53, 0x54, 0x55, 0x56, 0x57, 0x58, 0x59, 0x5a, 0x63, 0x64, 0x65, 0x66,
  0x67, 0x68, 0x69, 0x6a, 0x73, 0x74, 0x75, 0x76, 0x77, 0x78, 0x79, 0x7a, 0x83, 0x84, 0x85, 0x86,
  0x87, 0x88, 0x89, 0x8a, 0x92, 0x93, 0x94, 0x95, 0x96, 0x97, 0x98, 0x99, 0x9a, 0xa2, 0xa3, 0xa4,
  0xa5, 0xa6, 0xa7, 0xa8, 0xa9, 0xaa, 0xb2, 0xb3, 0xb4, 0xb5, 0xb6, 0xb7, 0xb8, 0xb9, 0xba, 0xc2,
  0xc3, 0xc4, 0xc5, 0xc6, 0xc7, 0xc8, 0xc9, 0xca, 0xd2, 0xd3, 0xd4, 0xd5, 0xd6, 0xd7, 0xd8, 0xd9,
  0xda, 0xe1, 0xe2, 0xe3, 0xe4, 0xe5, 0xe6, 0xe7, 0xe8, 0xe9, 0xea, 0xf1, 0xf2, 0xf3, 0xf4, 0xf5,
[... 674 lines cut ...]
    }
  }
  for ( k1 = 0; k1 < length_low1; k1++) {
    ch = myCAM.read_fifo();
    outFile.write((uint8_t*)&ch, 1);
    Serial.write(ch);
  }
  ch = (uint8_t) (0xFF);
  outFile.write((uint8_t*)&ch, 1);
  Serial.write(ch);
  ch = (uint8_t) (0xD9);
  outFile.write((uint8_t*)&ch, 1);
  Serial.write(ch);
  //Close the file
  outFile.close();

  //Clear the capture done flag
  myCAM.clear_fifo_flag();

  //Switch to LCD Mode
  myCAM.write_reg(ARDUCHIP_TIM, 0);
  Serial.println("file written");
  return;
}
`