ArduCAM/Arduino #459: Bad Bitmap Image

Davarco, 2019-11-27:
Hi, 
I'm attempting to send back bitmap images from the Arducam using ArduCAM_Mini_Video_Streaming.ino, and reading them using the Windows ArduCAM_Host_V2.exe. However, the images I get for SetToBitmap are bad, while the JPG setting works. Attached are the images. 
Here are the links: 
https://drive.google.com/open?id=1foKE-AVdfyVjPFvxMH7scYfLhqKV4pWD (jpg, good)
https://drive.google.com/open?id=1jnorS8a9MnaQWTpRmU26zzVL4IyPF8Km (bmp, bad)
For reference, I am interfacing the ArduCAM with an ESP32 (Adafruit Feather). 
Thanks!

UCTRONICS, 2019-11-27:
@Davarco,
It is due to the high and low bytes inversion. Just delete the first SPI.transfer() and try it again.

Bashima, 2020-03-06:
@Davarco can you please elaborate which spi.transfer() are you mentioning above? I am having similar issue.

OwensDataCenter, 2022-04-12:
I don't know the answer to that but that looks like the camera was going through an acid trip.