ArduCAM/Arduino #464: ArduCam Mini-5MP-Plus doesn't consistently capture images

STPell, 2019-12-12:
When using the ArduCAM_Mini_5MP_Plus_Multi_Capture2SD example with the ArduCam Mini-5MP-Plus, frames are not always retrieved. When the frames are not retrieved the program gets stuck on the start capture step (output below). Additionally, when retrieving the first round of frames is successfull the next capture attempt will almost always fail.

```SPI interface OK.
OV5642 detected.
SD Card detected.
start capture.
```

When I attempt to use the streamer example mentioned in  #446  (modified to allow the use of a 5MP-Plus) I get more consistent frame retrieval. However, the majority of requests are still failing.

I am powering the module with an external 3.3 V supply separate from the Teensy microcontroller driving the module.

ArducamTeam, 2019-12-12:
Hello,
It is possible that the camera connection is too long, causing the camera initialize failed.You can try again with a shorter cable.

STPell, 2019-12-12:
Hi.

Thank you for your quick response. I shortened the wires and I still have the issue, additionally the same wire setup worked for the 2MP version of the camera.

[image]

STPell, 2019-12-12:
I just chucked the camera on a ArduinoMega 2560 I have and powered it off of the 5 V regulator and it seems to work fine. Can the camera be run off of 3.3 or do I need to provide it a 5 V supply?

ArducamTeam, 2019-12-17:
Arducam mini 2MP and 5MP cameras support a voltage range of 3.3v to 5V.But notice that the voltage of your microcontroller should be  the same the camera.For example, ArduinoMega 2560 provides 5V, the camera needs to use 5V.

phillipblake, 2020-07-19:
Issue is not your setup, it is that the camera pulls more amps than the Arduino Uno is able to provide in a short amount of time.  The Arduino Uno is already low power device and expects any real power requirements to be satisfied by an external power source.  Power all the devices except the Arduino Uno from an external power supply and you should be good.