ArduCAM/Arduino #470: ArduCAM OV5642 on Due

gisellegk, 2020-02-07:
Hello, I'm trying to get the ArduCAM to work on the Arduino Due. I have MISO/MOSI/SCK connected to the SPI header in the middle of the board along with 5v and GND, and SDA and SCL of the i2c connections going to pins 20 and 21 of the Due. CS is pin 4.

I am running the ArduCAM_Mini_5MP_Plus_Video_Streaming example. 

I've added the following lines of code to help debug. 
```
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  Serial.print("temp: "); Serial.println(temp, HEX);
```
and 
```
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    Serial.print("vid: "); 
    Serial.print(vid, HEX);
    Serial.print(" pid: ");
    Serial.println(pid, HEX);
```

In running the I keep getting the error message 

```
ACK CMD ArduCAM Start! END
temp: 55
ACK CMD SPI interface OK. END
vid: 0 pid: 0
ACK CMD Can't find OV5642 module! END
```
(the last message repeats forever)
It appears to be able to communicate with the ArduCAM over SPI based on the test "temp" output but does not get a vid or pid upon request.

This exact code (only changing CS to 7 instead of 4)
On the Arduino Uno:
```
ACK CMD ArduCAM Start! END
temp: 55
ACK CMD SPI interface OK. END
vid: 56 pid: 42
ACK CMD OV5642 detected. END
```

What is going on?
I have already tried the fix suggested by [this post](https://www.arducam.com/arducam-arduino-due-board/), which doesn't work, but also doesn't make sense because I'm not using the SD card library (as far as I know) for this example sketch. Plus, the SPI bus does appear to be working.

UCTRONICS, 2020-02-11:
Hi,
It seems your i2c signal connection exist error. Please attach me your hardware connection and I will help you check it in detail.

gisellegk, 2020-02-11:
Hi,

I was originally using the default SDA / SCL pins (pins 20 and 21) and since you pointed out it could be an I2C error, I tried SDA1 and SCL1 and it worked. Thank you so much!

closed 2020-02-11 by gisellegk

giovanniguerrero3, 2020-02-17:
PSA: Want to make it clear for people trying to use the DUE that you have to do this on the memorysaver.h file:
[image]