ArduCAM/Arduino #471: "SPI Interface Error" when using mini 2mp plus, Uno, and hiletgo micro sd card adapter

Mziff, 2020-02-12:
Hello,

I have not been able to run ANY of the example codes for the mini 2mp plus, no matter what I use I get an SPI interface error. I am using an arduino uno and the hiletgo micro sd card adapter (which is also hooked up through SPI). I have the camera and the sd module hooked up to the same pins with the CS for the camera on 7 and the CS for the SD on 9 (according to the example code).  I have seen some suggestions that not all SD modules are able to work with the arducam, have you tested this particular model of micro SD adapter? Is there another way to wire the camera and SD module to the arduino uno board? I am following this [example](https://www.arducam.com/wp-content/uploads/2017/12/ArduinoConnection-.jpg) for wiring and connecting the SD modulue pins appropriately.

ArduCAM, 2020-02-13:
Yes, most of the SD module design is not following the SPI specification, and occupying the SPI bus all the time which cause the other SPI devices like Arducam module can not access the SPI bus correctly. You can use our examples without the SD card function to verify the Arducam hardware functionality first.

mgrehberg42, 2020-04-29:
I am having this same issue. I get an "SPI Interface error" immediately after uploading. If I disconnect the 5V line to the SD module, the SPI error will clear. A "SD Card Error" will then show as the SD card does not have power. After plugging back in the 5V line to the SD, the error will clear.

Serial:
10:55:35.756 -> SPI interface Error!
10:55:36.769 -> SPI interface OK.
10:55:36.769 -> ACK CMD OV2640 detected.
10:55:38.671 -> SD Card detected.
10:55:39.174 -> start capture.
10:55:39.174 -> CAM Capture Done.
10:55:39.174 -> capture total_time used (in miliseconds):1
10:55:39.174 -> The fifo length is :8388607
10:55:39.174 -> Over size.
10:55:39.174 -> save capture total_time used (in miliseconds):4
10:55:44.212 -> start capture.

The Serial will then pause at this second capture. From research on these forums the fifo length of 8388607 is not an accurate length.
Any suggestions?

UCTRONICS, 2020-05-02:
Please attache me your hardware connection. It seems the cable line is too long which will cause the signal very poor.