ArduCAM/Arduino #475: "SPI Interface Error"/"ACK CMD SPI interface Error! END" using Uno, Mini 2MP Plus, standard wiring, ArduCAM_Mini_2MP_Plus_functions

jamiemccoll1, 2020-03-06:
Setup:
Arduino UNO
ArduCAM 2MP Mini Plus
Wired as advised at https://www.arducam.com/wp-content/uploads/2017/12/ArduinoConnection-.jpg

Software:
ArduCAM_Mini_2MP_Plus_functions, also ArduCAM_Mini_2MP_Plus_VideoStreaming

Issue:
Getting an SPI error during setup, where temp is not being set to 0x55

ArduCAM_Mini_2MP_Plus_functions lines 62-64:
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);

COM Output:
ACK CMD ArduCAM Start! END
ACK CMD SPI interface Error! END
(repeats)

Host V2 Output:
 ArduCAM Start! 
 SPI interface Error! 
(repeats)

Have also tried with UNO's alternate ICSP ports, different CS pin in hardware + software (10), as well as with an Arduino Mega, all have the same issue.

Is something wrong with my ArduCAM?

BillyBee123, 2020-03-07:
I've had this error before while working with ArduCAM example code for my NodeMCU ESP8266 board.  I had to unplug the USB cable to fully power down the MCU, then close and re-open the Arduino IDE and ArduCAM_Host_V2.  After recompiling, this error resolved itself.

Otherwise, try running i2c_scanner.ino to see if you can read the I2C address of the ArduCAM.  If that doesn't work, you'll know there is a problem with your ArduCAM.

jamiemccoll1, 2020-03-07:
Hmm, restarting doesn't seem to help so far.

I tried running i2c_scanner.ino from https://gist.github.com/tfeldmann/5411375 with additional println's in the for loop. The code hangs after the first "error = Wire.endTransmission();" line testing address 0x01. When testing with a different I2C device (a magnetometer) however the i2c_scanner.ino code works no problem. 

Any ideas why this could be?

Sorry I'm still quite new to this, thank you for your help.

BillyBee123, 2020-03-07:
The i2c_scanner tests the I2C bus, rather than the SPI bus, but ArduCAM uses both.  If you can't read the I2C address of the ArduCAM, there's a problem with either your connections or the ArduCAM itself.

Try i2c_scanner again, but disconnect everything except GND, VCC, SDA, and SCL (the last two are the data and clock pins for the I2C bus).  If these four pins are connected correctly, but the I2C address of the ArduCAM can't be read, there's likely a problem with your ArduCAM.

jamiemccoll1, 2020-03-11:
Yeah, no response from the i2c_scanner with the ArduCAM unfortunately, working fine with other devices. Might have to try return the ArduCAM. 

Thank you for your time.

ArducamSupport, 2020-03-16:
@jamiemccoll1 
Hi，
Your COM output SPI Error, which is a problem of SPI bus, first confirm that the CS of the camera is 7, as shown in your wiring diagram.The second is to make sure that the SPI does not share a bus with other devices, because some SD module will not release the occupied bus.

jamiemccoll1, 2020-03-21:
The CS of the camera is 7 both in wiring and in software, and no other devices are connected to the Arduino.

ArducamSupport, 2020-03-21:
@jamiemccoll1 
Make sure you choose 'Arduino UNO' in the Arduino IDE, Tools-- >Board: because different development boards have different pin definitions, if you choose the development Board and the hardware are not consistent, there will be errors.

jamiemccoll1, 2020-03-22:
I have selected the Arduino Uno in the Tools section and made sure to be using the correct COM port also.