ArduCAM/Arduino #476: Replace ESP32-Cam with ArduCam OV2640

Emilex108, 2020-03-12:
Hi,

I bought an ESP32-Cam and have been using the WebServer script that comes with it to Stream the image from the camera to a locally hosted website. That being said, the quality isn't quite what I wanted it to be so I've been wanting to replace the camera that comes with the ESP32-Cam by my ArduCamOV2640. Problem is, I'm unsure how to wire it. Here is the pin layout currently used :

#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

Hopefully someone can shed some light into the wiring of it ! Thank you!

ArduCAM, 2020-03-14:
Please try this repo : https://github.com/ArduCAM/ArduCAM_ESP32S_UNO

BillyBee123, 2020-03-19:
Doesn't the ESP32-cam dev board also use an OV2640 camera?

ArducamSupport, 2020-03-20:
> Doesn't the ESP32-cam dev board also use an OV2640 camera?

ESP32-Cam uses the same ov2640 sensor as ArduCamOV2640 MINI, but ArduCamOV2640 MINI uses spi to transmit image data and uses fewer pins.