ArduCAM/Arduino #479: OV2640 - Send

beratyil, 2020-04-01:


beratyil, 2020-04-01:
Hi, 
I am trying to send image data with LoRa module from one Arduino to other. SPI interface of both of them is OK. I edited ArduCAM_Mini_2MP_OV2640_functions to send the image datas. As far as I understand temp helds the image datas. So I am sending temp variable in the read_fifo_burst to other Arduino. When I printed the temp I see last of meaningless data in sender's serial monitor and see nothing in the receiver's terminal. I changed Serial.println(temp) to Serial.write(temp) but there was nothing changed or I could not noticed. The output was meaningless as well. 
I would appreciate if you could help me

beratyil, 2020-04-01:
This is the code.
// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can test OV2640 functions
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <LoRa.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP platform.
#if !(defined OV2640_MINI_2MP)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
// set pin 7 as the slave select for the digital pot:
const int CS = 5;
const int SD_CS = 4;
const int cs_lora = 7;
int counter = 0;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
#if defined (OV2640_MINI_2MP)
  ArduCAM myCAM( OV2640, CS );
#else
  ArduCAM myCAM( OV5642, CS );
#endif
uint8_t read_fifo_burst(ArduCAM myCAM);
//-----------------------------------------------------
void setup() {
// put your setup code here, to run once:
uint8_t vid, pid;
uint8_t temp;
#if defined(__SAM3X8E__)
  Wire.begin();
  Serial.begin(9600);
#else
  Wire.begin();
  Serial.begin(115200);
#endif
Serial.println(F("ACK CMD ArduCAM Start! END"));

// set the CS as an output:
pinMode(cs_lora, OUTPUT);
digitalWrite(cs_lora, HIGH);
pinMode(CS, OUTPUT);
digitalWrite(CS, HIGH);

// initialize SPI:
SPI.begin();
  //Reset the CPLD
myCAM.write_reg(0x07, 0x80);
delay(100);
myCAM.write_reg(0x07, 0x00);
delay(100);
while(1){
[... 510 lines cut ...]
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.println(temp);
      delay(500);
      LoRa.println(temp);
      delay(500);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.println(F("ACK IMG END"));
      LoRa.println(F("ACK IMG END"));
      Serial.write(temp_last);
      LoRa.write(temp_last);
      
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}

ArducamSupport, 2020-04-02:
Hi,
I recommend that you run the demo with Arduino and camera to make sure there are no problems with the camera.Then print the data to you want to send in your code to the serial port, make sure the data is correct, and finally replace the print function with the function sent to another device.

beratyil, 2020-04-02:
I checked whether camera works normally or not. Camera works just fine.
Image data is held in temp, isnt it? So i have to print temp in demo?

ArducamSupport, 2020-04-02:
Hello,
`temp = SPI.transfer(0x00);`
This temp is the image data read from FIFO. 0XFF and 0XD8 is the head of JPEG image. 0XFF and 0XD9 is tail of JPEG image.Print the data between 0XFF,0XD8 and 0XFF, 0XD9. Make sure the image data is right.

beratyil, 2020-04-04:
As your recommend I am trying to write the image data to serial monitor only. In demo code read_fifo_burst function is given like that. ArduCAM App doesnt show values of temp or temp_last. This is why I am using Arduino's Serial Monitor. And in that serial monitor, I cant see any data unless I changed serial.write(temp) to serial.print(temp). 

uint8_t read_fifo_burst(ArduCAM myCAM)
{
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  length = myCAM.read_fifo_length();
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //512 kb
  {
    Serial.println(F("ACK CMD Over size. END"));
    return 0;
  }
  if (length == 0 ) //0 kb
  {
    Serial.println(F("ACK CMD Size is 0. END"));
    return 0;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  temp =  SPI.transfer(0x00);
  length --;
  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.println(F("ACK IMG END"));
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}

beratyil, 2020-04-04:
[image: image.png]
This is the output of example code ArduCAM_Mini_2MP_OV2640_functions.
Should I use your app?If so, what should I do to see value of temp?

Arducam Support <notifications@github.com>, 2 Nis 2020 Per, 16:53 tarihinde
şunu yazdı:

ArducamSupport, 2020-04-04:
Hello,
We asked you to print the  values between 0XFF 0XD8 and X0FF 0XD9 to make sure the image data is correct. The printe function you can just use `Serial.println` directly, and the data will be output to  the Arduino‘s Serial port monitor.If there is noting wrong with the hardware that the image data will be right.You can use this [demo](https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_2MP_OV2640_functions) that can be displayed by host app.App's download [address](https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows/ArduCAM_Host_V2.exe) Make sure the baud rate of the software is the same as set in the code.
If still not up to your requirements, please send your complete operation video and code through the file way to our mailbox.

beratyil, 2020-04-08:
> Arducam Support <notifications@github.com>, 4 Nis 2020 Cmt, 18:09
> tarihinde şunu yazdı:

I sent you the video of what I am doing.
As you can see, in the first part, it takes a picture without a problem. In
both 320x240 and 1600x1200 resolutions, camera took an image as it supposed
to be.
In 00.52, I opened the Serial monitor of Arduino and sent 0x10. It is for a
single shoot and I want to use my camera in this format. Output was not
meaningful for a human. I tried to print output with Serial.print and
nothing was changed so I left as Serial.write.
Then in 01.57 I opened another code.This code is essentially same
with ArduCAM_Mini_2MP_OV2640_functions but I thought when Arduino reads the
data from SPI, a problem occurs because of the baudrate of the Serial
communication with PC. Hence I declared an array before the while loop
named as TheArray. I sent all the image data read from SPI to this array
and when this loop finishes, another loop begins. So i am printing the
image data in this loop. But the output as blended as in the first part.
Only part I changed is read_fifo_burst in
ArduCAM_Mini_2MP_OV2640_functions, so i put here, only this function. What
am I missing. You said if there was no mistake in the hardware, I have to
see image data without problem in serial monitor of arduino. So what am i
missing?
uint8_t read_fifo_burst(ArduCAM myCAM)
{
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  uint32_t length2 = 0;
  length = myCAM.read_fifo_length();
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //512 kb
  {
    Serial.println(F("ACK CMD Over size. END"));
    return 0;
  }
  if (length == 0 ) //0 kb
  {
    Serial.println(F("ACK CMD Size is 0. END"));
    return 0;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  temp =  SPI.transfer(0x00);
  length --;
  length2 = length;
  Serial.begin(921600);
  uint8_t TheArray[length];
  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      //Serial.print(temp,HEX);
      //Serial.println();

    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      /*Serial.println(F("ACK CMD IMG2 END"));
      Serial.print("temp_last = ");
      Serial.print(temp_last,HEX);
      Serial.print(" temp = ");
      Serial.print(temp,HEX);
      Serial.println();*/
      TheArray[length] = temp;
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break
while,
    break;
    delayMicroseconds(15);
  }
  Serial.println(F("ACK CMD IMG1 END"));
  myCAM.CS_HIGH();
  uint32_t i;
  for (i=0; i<= length; i++)
  {
    Serial.println();
    Serial.println(TheArray[i],HEX);
  }
  is_header = false;
  return 1;
}

Ab ylmz <adnanberatyilmaz@gmail.com>, 8 Nis 2020 Çar, 17:54 tarihinde şunu
yazdı:

ArducamSupport, 2020-04-09:
Thank you very much for the detailed video.
Firstly, explain why the output to the Serial monitor using `Serial.print()` was not
meaningful for a human.`Serial.print()` is print data to the Serial port as human-reading ASCII text, and `Serial.write()` is write binary data to the Serial port.The serial port monitor that comes with Arduino displays the ASCII code directly. You can use a separate serial port tool to see the output data.Serial.println() is generally used to output temp values that are normally displayed in the Arduino Serial port, but there are minor problems with this program that cause Serial.println() to fail.
Comment out:
```
if (Serial. available())
{
...
}  
```
and then 
```
mode == 1
start_capture =1;
```
You can verify this problem. We will optimize this problem in later versions.
You define an array and then output, which is not a good idea, because the image data is so large that it exceeds the Arduino ram, and the program has an exception.

halukmy, 2021-03-08:
has anyone tried ? is it working?