ArduCAM/Arduino #480: error: ‘i2c_smbus_write_word_data’ was not declared in this scope

thaiwash, 2020-04-13:
I tried to raspberry this, and i went to ArduCAM/examples/RaspberryPi and typed make

g++ -std=c++0x -I./ -I../../../ArduCAM -c arducam_arch_raspberrypi.c
arducam_arch_raspberrypi.c: In function ‘uint8_t arducam_i2c_word_write(uint16_t, uint8_t)’:
arducam_arch_raspberrypi.c:125:3: error: ‘i2c_smbus_write_word_data’ was not declared in this scope
   i2c_smbus_write_word_data(FD, reg_H, value);
   ^~~~~~~~~~~~~~~~~~~~~~~~~
arducam_arch_raspberrypi.c:125:3: note: suggested alternative: ‘i2c_smbus_ioctl_data’
   i2c_smbus_write_word_data(FD, reg_H, value);
   ^~~~~~~~~~~~~~~~~~~~~~~~~
   i2c_smbus_ioctl_data
arducam_arch_raspberrypi.c: In function ‘uint8_t arducam_i2c_word_read(uint16_t, uint8_t*)’:
arducam_arch_raspberrypi.c:140:7: error: ‘i2c_smbus_write_byte_data’ was not declared in this scope
   r = i2c_smbus_write_byte_data(FD,reg_H,reg_L);
       ^~~~~~~~~~~~~~~~~~~~~~~~~
arducam_arch_raspberrypi.c:140:7: note: suggested alternative: ‘i2c_smbus_ioctl_data’
   r = i2c_smbus_write_byte_data(FD,reg_H,reg_L);
       ^~~~~~~~~~~~~~~~~~~~~~~~~
       i2c_smbus_ioctl_data
arducam_arch_raspberrypi.c:143:13: error: ‘i2c_smbus_read_byte’ was not declared in this scope
   *regDat = i2c_smbus_read_byte(FD);
             ^~~~~~~~~~~~~~~~~~~
arducam_arch_raspberrypi.c:143:13: note: suggested alternative: ‘i2c_smbus_data’
   *regDat = i2c_smbus_read_byte(FD);
             ^~~~~~~~~~~~~~~~~~~
             i2c_smbus_data

thaiwash, 2020-04-13:
I added #include <smbus.h>

to arducam_arch_raspberrypi.c

and copypasted the content of the .c file from
https://stackoverflow.com/questions/14268530/linking-smbus-c-to-smbus-h
(I know, hackish)

I got it compiling! 

now I need to figure out how to plug the pins, I know how to wire 4 wires now, 3.3v, gnd and SCL and SCK

running: sudo ./ov2640_capture -c test.jpg 320x240
I get: Unable to open I2C device: No such file or directory

ArducamSupport, 2020-04-14:
Hi,
You may not turned on the I2C device, so you can try this [link](https://www.raspberrypi.org/forums/viewtopic.php?t=115080).

thaiwash, 2020-04-14:
I have i2c working since i can run my gyro no poblem. I'm using only 4 pins right now, 3.3v, gnd and sck sda its not even detecting the camera with i2cdetect -y 1

ArducamSupport, 2020-04-15:
Perhaps you have a problem with the I2C method, I suggest you refer to our example on raspberry pi.
[](https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera)

thaiwash, 2020-04-15:
> 
> 
> Perhaps you have a problem with the I2C method, I suggest you refer to our example on raspberry pi.
> [Link](https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera)

Okay, i'll try that, what about the pin configurations, is having only sda and sck okay for spi?

ArducamSupport, 2020-04-15:
The I2C pins of the camera are SCL and SDA.
The SPI pins are CS、MOSI、MISO and SCK, is your I2C clock using SCK?

thaiwash, 2020-04-15:
no, its using SCL

thaiwash, 2020-04-15:
So, I wire up both I2C and SPI before running the program?

thaiwash, 2020-04-15:
https://pinout.xyz/pinout/spi#

thaiwash, 2020-04-15:
so,
pins are 
BCM9   = 21 (MISO)
BCM10 = 19 (MOSI)
BCM11 = 23 (SCLK)
 BCM0  = 27 (SPI0)

I'm going to wire up SPI only.

ArducamSupport, 2020-04-15:
> so,
> pins are
> BCM9 = 21 (MISO)
> BCM10 = 19 (MOSI)
> BCM11 = 23 (SCLK)
> BCM0 = 27 (SPI0)
> 
> I'm going to wire up SPI only.

BCM0 is not used, please wire the CS to BCM17.
You could see the [README](https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera).

thaiwash, 2020-04-15:
This is from the readme,

> //SPI pins 
> //set 9 (MISO), 10 (MOSI), 11 (SCLK) alt fxn 0 (SPI0)

I dont know what alt fxn means thoough..

UCTRONICS, 2020-04-15:
Sorry for the readme mislead you. Just ignore the alt fxn (SPI0), which just want to tell you to use SPI0.

thaiwash, 2020-04-15:
[image]
Lets draw a picture here, where is MISO MOSI SCLK and SPI0 on the camera?

ArducamSupport, 2020-04-16:
The [ArduCAM/RaspberryPi/SPI_Camera/](https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera) is used for the specified camera, as shown in the following figure, for reference: [https://www.arducam.com/product/arducam-2mp-spi-camera-b0067-arduino/](https://www.arducam.com/product/arducam-2mp-spi-camera-b0067-arduino/)
Not the parallel camera in your picture.
[image]

phymatld, 2020-08-08:
I have the same question at first。finally ，I can compile it.
but when I run the command line **sudo ./ov5642_capture -c test.jpg 320x240**
I get  the result as below

**OV5642 detected.
Changed resolution1 to 320x240
Start capture
CAM Capture Done
Reading FIFO and saving IMG
The length is 2792**

but ,the size of picture is 0

WHY?

ArducamSupport, 2020-08-09:
Hi,
We noticed that there is no clearing FIFO in your output. Please clear FIFO before you start capture. In addition, you can print the data read from FIFO to the terminal. If the data is jpeg data, then there may be a problem with the way you save the image.

phymatld, 2020-08-09:
thanks .but how to clear FIFO?
in fact ,I bought a ARDUCAM OV5642 and raspberry pi 4B ,I want to use the camera to take some photo or video.but I didn`t find any tutorial to do that ,so I use this lib to try

ArducamSupport, 2020-08-09:
Are you using SPI camera on Raspberry Pi? If yes, please refer to [https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera](https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera)

phymatld, 2020-08-09:
of cause I try it  before .
but when run the command line **sudo ./run_Arducam_Demo**
But I can't get any results, and the program won't report an error.

ArducamSupport, 2020-08-10:
Hi,
For PI4, please execute the following command to update the wiringpi library.
[image]

Then you need to make the following changes in bcm283_board_driver.h:
[image]

Finally, recompile and execute.

phymatld, 2020-08-10:
[image]

according to your advise，I install the latest version of wiringpi，and change the bcm283_board_driver.h file 
But there's a problem.

[image]
I connect the ARDUCAM ov5642 and pi4 according tothe picture。
[image]

ArducamSupport, 2020-08-10:
Hi,
Please refer to this connection.
[image]

phymatld, 2020-08-10:
[image]
finally，it works！
but，my mobile can not connect to the camera，
I have check the IP and port 。
my mobile status is always connecting...

phymatld, 2020-08-10:
[image]

[image]

here is the screenshot

ArducamSupport, 2020-08-10:
Hi,
Due to our negligence, there is a problem with the latest version code. Please use the code of this version. The part that needs to be modified has been explained in the reply above.
[https://github.com/ArduCAM/RaspberryPi/tree/9ebf547e2795f81835d0490a957aaad6e70e6595/SPI_Camera](https://github.com/ArduCAM/RaspberryPi/tree/9ebf547e2795f81835d0490a957aaad6e70e6595/SPI_Camera)

[image]

phymatld, 2020-08-10:
I am sorry but I really know how to download this version of demo.
can you show me the command?

phymatld, 2020-08-11:
thanks for your help. 
I can use it now .
but I still have a question ,can I use it only by mobile?
can I save photoes on raspberry derecctly?

ArducamSupport, 2020-08-11:
Hi，

You can save images directly to the Raspberry Pi, but there is no example of directly saving images on our Raspberry Pi. We suggest you refer to the Raspberry Pi code and the Arduino version code for testing.
We are very busy lately, sorry for not being able to write a detailed example for you.

phymatld, 2020-08-11:
in fact, I got the code by  **git clone https://github.com/ArduCAM/Arduino.git**
I told you that I capture a photo but the size is zero
now ,I change the connection according to the picture you show ,now I got  a error show as below

SPI interface error!

ArducamSupport, 2020-08-12:
Hello,
The repository you cloned above is an example of using the SPI camera on the Arduino platform. You can refer to the process of taking pictures and saving images, which cannot be used directly on the Raspberry Pi.
Now that you have run through our reference example on the Raspberry Pi, I suggest you modify it on the run through example.
In addition, the SPI interface error is either a connection error or a problem with the spi driver.

DhimanAirao, 2025-02-17:
I am facing same compilation error, can someone explain how to solve it?