ArduCAM/Arduino #483: Trigger an image capture from another arduino module

Kel2D2, 2020-04-22:
Hi, 

I'm doing a project for school - I have little coding experience so please forgive lack of proper terminology, but is there a way to trigger an image capture on an arducam from data from another arduino module? 

Specifically, I have a pulse sensor and an adurcam OV2640 2mp module wired to my arduino uno r3 board. The pulse sensor reads my heartbeat in bpms - when my bmp reaches a certain rate (say 110 bpm, can i use that data to trigger the arducam to take a picture?

ArducamSupport, 2020-04-23:
Hello,
I'm glad to help you. The one way is that call the Capture function and read out the image data when the bmp reaches a certain rate. The another way is that connect one pin to the external trigger pin on the camera and control it when the bmp reaches a certain rate.

Kel2D2, 2020-04-24:
Thank you for you reply. 
Where can I find that 'capture function' in my code and how do I edit that? I have searched through the Arducam site, documentation, and tutorials and it's not clear at all. Right now I am using the example code for the OV2640 functions, and I have added in my code for the pulsesensor setup. (see below)

I have been reading through various threads and help forums and I'm finding out that a lot of people have asked about how to automatically trigger an image with an arducam, but there isn't a clear way to do that. Can you provide some clarity on that question? 

Here is the code I am using: 

//PulseSensor
#define USE_ARDUINO_INTERRUPTS true    // Set-up low-level interrupts for most acurate BPM math.
#include <PulseSensorPlayground.h>     // Includes the PulseSensorPlayground Library.
//PulseSensor Variables
const int PulseWire = 0;       // PulseSensor PURPLE WIRE connected to ANALOG PIN 0
int Threshold = 550;           // Determine which Signal to "count as a beat" and which to ignore.                            
PulseSensorPlayground pulseSensor;  // Creates an instance of the PulseSensorPlayground object called "pulseSensor"
// ArduCAM Mini
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP platform.
#if !(defined OV2640_MINI_2MP)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
#if defined (OV2640_MINI_2MP)
  ArduCAM myCAM( OV2640, CS );
#else
  ArduCAM myCAM( OV5642, CS );
#endif
uint8_t read_fifo_burst(ArduCAM myCAM);

void setup() {
//PulseSensor Setup
pulseSensor.analogInput(PulseWire);
pulseSensor.setThreshold(Threshold);

//Camera Setup
uint8_t vid, pid;
uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
  Serial.begin(115200);
#else
  Wire.begin();
  Serial.begin(921600);
#endif
Serial.println(F("ACK CMD ArduCAM Start! END"));
// set the CS as an output:
pinMode(CS, OUTPUT);
digitalWrite(CS, HIGH);
// initialize SPI:
SPI.begin();
  //Reset the CPLD
myCAM.write_reg(0x07, 0x80);
delay(100);
myCAM.write_reg(0x07, 0x00);
delay(100);
[... 470 lines cut ...]
  temp =  SPI.transfer(0x00);
  length --;
  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.println(F("ACK IMG END"));
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}

ArducamSupport, 2020-04-24:
Hi,
This is the capture process:
`myCAM.flush_fifo();
myCAM.clear_fifo_flag();
//Start capture
myCAM.start_capture();

if (myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK))
{
Serial.println(F("ACK CMD CAM Capture Done. END"));delay(50);
read_fifo_burst(myCAM);
//Clear the capture done flag
myCAM.clear_fifo_flag();
}`
After capture done, you can read out the data from fifo.
The above demo due to the receive function of serial port , it's not too clear.
For detail process of capture and store, I suggest you refer to this[https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_PIR_Trig_Capture2SD](https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_PIR_Trig_Capture2SD ).