ArduCAM/Arduino #484: SPI/SD Errors ArduCAM_Mini_2MP_TimeElapse2SD

mgrehberg42, 2020-04-29:
I'm using the example referenced in the issue title. My SD card module is connected to the same MOSI, MISO, and SCK as the Arducam 2MP Plus. The SD_CS is pin 9, the Cam_CS is 7. 

The Serial prints for my code is below:

ArduCAM Start!
SPI interface Error!
SPI interface Error!
SPI interface Error!
SPI interface Error! _-----At this point I disconnected the 5V line from the SD module. I've found out this triggers the SPI interface for the camera to be detected_

SPI interface OK.
ACK CMD OV2640 detected.
SD Card detected. _------ I put the 5V line back into the SD module so it could be read._

start capture.
CAM Capture Done.
capture total_time used (in miliseconds):1
The fifo length is :8388607 __-----From my research on the forums, this value for a fifo length is an_ invalid value_

Over size.
save capture total_time used (in miliseconds):4

_----Code crashes before trying to open or save the file_

start capture.

_----Code pauses here until I disconnect the 5V line from the SD module again_

CAM Capture Done.
capture total_time used (in miliseconds):-16018
The fifo length is :271368

_----While the SD is disconnected the Camera gives a valid fifo length reading. However, since there is no supply to the SD module the file saving will not work_

File open failed

It seems to me there is an issue with having multiple slaves on the same loop for this code. 
Do I have to write the SD_CS low for it to activate while the camera is trying to save? 
Why does the Camera not respond when the SD module has power?
The SD module I am using is a HiLetGo SD Module. I've tested both independently and they work.

ArducamSupport, 2020-04-30:
Hi，
Due to the poor design of some SD card slots, the SPI bus is not released, which cause the bus can't connect multi device. So I advise you change another SD card slot.

rehrehreh, 2020-04-30:
Thanks for the reply!
Is there an SD module that you can recommend for use with Arducam?
Do you think it would be possible for me to use a switch or transistor to release the SPI bus of the poor SD module that I have?

ArducamSupport, 2020-05-01:
Hi,
I advise you use the Arducam Arduino UNO board, we don’t have a single SD module.
I think it is possible that use a switch or transistor to release the SPI bus of the poor SD module.

just4give, 2020-12-07:
Hi,
I am facing exact same problem. I need to use Arduino Nano BLE sense with ArducCam and SD card because of in-built microphone. I am using [HiLetGo](https://www.amazon.com/gp/product/B07BJ2P6X6) SD card module. Any suggestion to resolve the issue? I read above about a switch but not sure how to use it.