ArduCAM/Arduino #485: ArduCam Mini 5MP with Arduino MKR FOX 1200

dansebastian1, 2020-05-21:
I am trying to connect the Arducam with the MKR FOX 1200 (to capture image and save it to SD card) but I encounter the SPI Interface error. I have found a couple of workarounds for the MKR1000, which include adding a block of code in the ArduCAM.h file, but I cannot get any result with my board. Could you please provide support or advice for this?

I have also tried various names like "MKR1200", "MKRFOX1200" etc. but, just as expected, they did not work.

ArducamSupport, 2020-05-25:
Hi，
We haven't specifically tested the Arduino MKR FOX 1200 yet, but you can refer to this [link](https://www.arducam.com/forums/topic/arducam-mini-5mpov5642-with-arduino-mkr-1400-board-results-in-very-dark-images/).

dansebastian1, 2020-05-28:
Hello,
I tested the Mini_Capture2SD and Mini_Video2SD but I keep getting "SPI Interface Error" on whichever CS pin I connect my Arducam, ranging from 0 to 7. I always modify the SPI_CS in the code when I change the pins. Also, when I use pin 6 and try to upload, the board shuts off completely.

ArduCAM, 2020-06-01:
Hello,
Due to we don't have Arduino MKR FOX 1200 develop board, we can't help you test it now.
I have some suggestions for you. Using a new board, before testing the spi camera, you will 
ensure if the SPI works normal. You can test it using the official demo Arduino provided.
After ensure the spi works normally. Let's move on our SPI camera.