ArduCAM/Arduino #489: "SPI interface Error": Arducam Mini 5MP Plus OV5642 with Arduino Nano 33 Ble

DenVys, 2020-06-12:
I have an **Arduino Nano 33 Ble**. This board supports either SPI and I2C interfaces.
Also I have **Arducam Mini 5MP Plus OV5642**.
I connected the camera to the board and follow one of the examples (_ArduCAM_Mini_5MP_Plus_OV5642_RAW_)
But in the result I get ```"SPI interface Error"```

Piece of code that causes the problem.
```
while(1){
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
    Serial.println(F("SPI interface Error!"));
    delay(1000);continue;
  }else{
    Serial.println(F("SPI interface OK."));break;
  }
}
```

Any idea what am I doing wrong?
I double checked my board and all connections. Everything looks fine. But still I have this error. 
Is it soft problem ?
Or is it hardware problem?
In which direction should I dig to find the solution? Any advice?

DenVys, 2020-06-14:
It looks like that **Arducam Mini 5MP Plus OV5642** has 8MHz SCLK speed. While my board has 64MHz clock speed. Which is a little bit higher than I need 🙂
I think it's a root of the problem.

closed 2020-06-14 by DenVys

orionparrott, 2020-06-16:
hmm, board (processor) clock speed is different from SCLK which is the clock speed of the SPI bus. But you clued me in to the fact that perhaps I need to set my SPI clock speed different for the OV2640 (Arducam Mini 2MP Plus) camera that I'm trying to use. Will update.

DenVys, 2020-06-16:
I found this information in the hardware specification to my camera module. 
For example, [here](https://arducam.com/downloads/shields/ArduCAM_Mini_5MP_Plus_OV5642_Camera_Module_Hardware_Application_Note.pdf) for **Arducam Mini 5MP Plus OV5642** noted that _The stableSCLK speed is 8MHz, care should be taken whenover clock the SPI bus speed._
So, I think it will be nice to check that in the documention.

orionparrott, 2020-06-16:
Looks like Arduino SCLK can be set with 
`  SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0)); //OWP added testing`

which would make it match the OV5642 SPI clock speed you require. My understanding is you can set SPI clock to 8 MHz as long as your system (CPU) clock speed is a multiple of 8 MHz.

DenVys, 2020-06-17:
After other few days I found this [issue](https://github.com/ArduCAM/Arduino/issues/442) where there are images how to connect Arduino Nano 33 BLE Sense to camera module. So, I went back to my initial setup and I realized that putting 3v pin as VCC fixed the problem.

Anyway, my problem was misconfigured VCC port in the first place. I set +3v3 pin as VCC in my camera and now everything works fine.

nandoos97, 2024-07-04:
I am facing the same problem with OV2640 (Arducam Mini 2MP Plus) camera.Any idea how to solve?