ArduCAM/Arduino #490: ArduCam_Host_V2.0_Mac and Arducam Mini-2MP-Plus unreliable single shot over serial

orionparrott, 2020-06-16:
I'm using Arducam Mini-2MP-Plus (OV2640) and trying to send images over serial to ArduCam_Host_V2.0_Mac on Catalina (10.15.5). 
I'm using Arduino Nano 33 BLE Sense. 
Running example script ArduCAM_Mini_Video_Streaming.ino, without modification.

The images are sent and saved on the Mac and I can save as BMP or JPG.

BUT the connection is unreliable and the image is saved only about 20% of the time. I cannot figure out what is needed to make the image capture happen every time.
Sometimes when I click capture, the image that is saved is 3 images behind, tested by holding up different counting fingers for each attempted photo capture. 

Why can it work some times but not all the time? I have baud rate set to 921600. 

Thank you!

DenVys, 2020-06-16:
Hey
I'm using Arduino Nano 33 BLE Sense too but with different camera module. I found out that my camera module can't working with Arduino Nano 33 BLE Sense because that board has too high clock frequency. I wrote it here #489 
Maybe somehow it's the same problem that you have ?

orionparrott, 2020-06-16:
Hi, thanks for your reply. It could be related. Looking further. Replied to your post on your thread. SCLK and processor clock speed are 2 different things.

orionparrott, 2020-06-16:
Looks like SCLK can be set with 
`  SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0)); //OWP added testing`

Thanks for pointing me in that direction.

ArducamSupport, 2020-06-22:
Hi,
Did you connect the Arduino Nano 33 BLE to the camera via the dupont wire?
The length of the line on the impact of serial port signal is particularly large, I recommended to try a short line.If it still doesn't work, slow down the SPI speed to test.