ArduCAM/Arduino #494: Very low power mode.

astetyn, 2020-07-24:
Hello.

I want to achieve very low current on my ArduCam 2MP mini plus.

I tried basic low power mode with `set_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);` but that was not enough. I experimented with registers 0x83 (bit[6]) and 0x86 (bit[2]) as mentioned [here](https://www.mpja.com/download/33275mpsoftware%20application.pdf) but there was no difference between this and the first low power method. (register 0x83 did nothing)
With both methods I cut current from ~60mA to ~30mA.

Is it possible to achieve current consuption below e.g. 5mA?

ArducamSupport, 2020-07-26:
Hi,
Using the method you mentioned above just puts the camera sensor into low power consumption mode, but CPLD and FIFO still have power, so if you want the current to be low enough, you can turn on or off the camera by controlling the power pin of the camera module. But in this way you need to re-initialize the camera every time you turn on the camera.

astetyn, 2020-07-29:
Hello,

Thanks for the answer.
Re-initialization is not a problem for me. Bigger problem I am worrying about with turning camera off is unexpected behaviour of SPI and I2C pins because there will be other devices on these buses still active and communicating.

ArduCAM, 2020-07-31:
Can you do that testing on your end?

astetyn, 2020-07-31:
Yes. I did tests today. I am designing a new PCB (which will contain ArduCam) so I do not know what will be the behaviour there but I tried similar scenario with ATMega1284P as a I2C master.
ArduCam is the first slave and MCU9250 is the second slave on I2C. Both devices were working until I unplugged GND and VDD pin from ArduCam and plugged back, I2C error 4 was thrown. When I unplugged just VDD pin, the I2C bus was working. GND pin should be plugged constantly so this is not the issue and both I2C bus and SPI bus should be working even with VDD not connected to ArduCam.
After tests I can say that this can be closed because unplugging VDD pin is the only way how to achieve low power consuption and it should not cause problems on I2C and SPI bus. I will do some more tests for possible current leaks over both buses.

closed 2020-08-01 by astetyn

ArduCAM, 2020-08-01:
Thanks for your effort and report.