ArduCAM/Arduino #496: Connecting OV2640 camera to Arduino BLE 33

gigwegbe, 2020-08-11:
I recently got the [OV2640 camera]( https://alexnld.com/product/xd-95-ov2640-camera-module-200w-pixel-stm32f4-driver-support-jpeg-output-for-arduino/), and there is no documentation on how to connect it to Arduino BLE 33/Arduino Nano. I will need the wiring connection and any other useful link. Thanks

ArducamSupport, 2020-08-12:
Hi,
The camera we use on arduino is spi interface, as shown below.
[image]

The camera you purchased is not our product, please contact the dealer where you purchased the product.

ArducamSupport, 2020-08-12:
This is the product link of our spi camera：[https://www.arducam.com/docs/spi-cameras-for-arduino/](https://www.arducam.com/docs/spi-cameras-for-arduino/)
Hope it helps you.

closed 2020-08-12 by gigwegbe