ArduCAM/Arduino #498: Spresense board

Blakesley1985, 2020-08-30:
Does anyone know if the Sony spresense board is supported?  When I try to run it it gives me a couple errors with regtype and regsize.  The board itself is supposed to run most traditional arduino librarys.

ArducamSupport, 2020-08-31:
Hi，
We don't have Spresense board, so we can't test it for you, but you can paste the error message and we will try our best to help you analyze the problem.

Blakesley1985, 2020-08-31:
Arduino: 1.8.13 (Windows 10), Board: "Spresense, MainCore, 768 KB (Default), Disabled, 115200"

In file included from C:\Users\Landon Work\Documents\Arduino\libraries\ArduCAM\examples\mini\ArduCAM_Mini_Video_Streaming\ArduCAM_Mini_Video_Streaming.ino:21:0:

C:\Users\Landon Work\Documents\Arduino\libraries\ArduCAM/ArduCAM.h:768:2: error: 'regtype' does not name a type

  regtype *P_CS;

  ^

C:\Users\Landon Work\Documents\Arduino\libraries\ArduCAM/ArduCAM.h:769:2: error: 'regsize' does not name a type

  regsize B_CS;

  ^

exit status 1

Error compiling for board Spresense.

ArducamSupport, 2020-08-31:
Hi，
We have modified the library file to compile and pass your above demo, but since there is no board, it is up to you to test it. What needs to be reminded is that please pay attention to whether the SPI pins of the board are consistent with those used in the demo.
The file is as follows.
[ArduCAM.zip](https://github.com/ArduCAM/Arduino/files/5149192/ArduCAM.zip)

Blakesley1985, 2020-09-01:
hello,
i've got the updated library files and it compiles correctly but i get a
spi interface error.  I did a serial print and got a value of 255 instead
of 0x55 in the temp value.  Does it store this value in eeprom?

ArducamSupport, 2020-09-01:
Hi,
 Please check whether the SPI pins of the board are consistent with those used in the demo.

Blakesley1985, 2020-09-02:
i double checked the connection from the demo,  connections are the same as
they are on the arduino uno Rev3.  The demo runs on the uno but gives me a
spi error on the spresense board.  I really appreciate all of the help with
trying to get this to run.

ArducamSupport, 2020-09-03:
Hi,
The pin definition between arduino uno and spresense board is different, you should accroding to the real SPI and I2C pin to change the wiring connection.You may also need to change the SPI CS pin. You can find the pins of spresense board in this directory:C:\Users\YOUR PC\AppData\Local\Arduino15\packages\SPRESENSE\hardware\spresense\2.0.1\variants\spresense

Blakesley1985, 2020-09-04:
[image: 20200903_121425.jpg][image: 20200903_121443.jpg]
I have checked the pinouts on both the uno and spresense extension board.
I have also added pictures of the current wire setup. The CS pin is 7 and I
changed
 that in the software.  I also tested another library and a tft screen that
uses miso and mosi.  The screen works with the extension board, it does not
send information to the computer, however. Any other ideas that would cause
an error with the spi or see an error in the wiring that i have overlooked?

ArducamSupport, 2020-09-04:
Hi,
Your picture upload may have failed, I can’t see your picture.
If the connection is correct, the main reasons for the SPI error as are follows: 
1. The SPI clock speed is too fast.
2. The SPI connection line is too long.
3. The camera is not powered.

Blakesley1985, 2020-09-04:
[image]
[image]
Is there a way to adjust the spi timing to check to see if that is the issue?  I'm using short jumper wires that came with the camera.

ArducamSupport, 2020-09-04:
Hi,
Do you mean that the TFT screen has been successfully driven using the same SPI line?
This seems to indicate that there is no problem with the SPI line. You can refer to the SPI library corresponding to Spresense to modify the SPI.
We don't have a Spresense board, I'm sorry I can't help you test it.

Blakesley1985, 2020-09-04:
I have a RA8875 tft driver board that i was able to get to work on the
spresense board.  So what i need to do is modify the library files for the
camera to work like the other libraries that do work?

ArducamSupport, 2020-09-04:
Hi，
You can first use the SPI method in the RA8875 library to read and write the camera's SPI register, if the read and write succeeds. Then you can refer to the RA8875 library to modify the SPI method in our library.