ArduCAM/Arduino #501: "SPI interface error!" With OV5642 plus and raspberry pi 3 b+

NateDKing, 2020-10-12:
I have followed the readme.md instructions with this library. When I run "sudo ./ov5642_capture -c test.jpg 320x240" as shown in the raspberry pi example folders readme, it shows the error code "SPI interface error!" and does nothing. 

I found this thread (https://github.com/ArduCAM/Arduino/issues/480), which was able to solve some troubles that I had earlier. But that thread also had a link to a repository (https://github.com/ArduCAM/RaspberryPi/tree/9ebf547e2795f81835d0490a957aaad6e70e6595/SPI_Camera) that I downloaded, and when I run it, it gives me the error of "Waitting connection..." or "Error on binding: Address already in use".

Current wiring:
CS: 24 (I have also tested it with both programs on pin 7)
MOSI: 19
MISO: 21
SCK: 23
GND: 9
VCC: 1
SDA: 3 
SCL: 5

ArducamSupport, 2020-10-13:
Hi,
When you use the SPI camera on raspberry, please refer to this link: [https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/RaspberryPi](https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/RaspberryPi).
Please ignore this link: [https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/RaspberryPi](https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/RaspberryPi).
If there are still problems after the test, please feel free to contact us.

NateDKing, 2020-10-14:
Thanks for the correct version of the code. I ran it on the raspberry pi, and It still seems to be getting stuck on `Waitting connection...` which is one of the same errors as I got before.

This was the log after I ran the command: `sudo ./run/Arducam_Demo`
`SPI1 interface OK!` 
`Cant find OV2640 module!`
`Cant find OV5640 module!`
`OV5642 detected.`
`the format is JPEG`
`set resolution OK`
`Waitting connection...`

I have enabled the following in boot/config.txt:
`dtparam=i2c_arm=on`
`dtparam=i2a=on`
`dtparam=spi=on`

and I also currently have the cs pin in BCM 17.

Any other suggestions?

ArducamSupport, 2020-10-15:
Hi,
Displaying 
`Waitting connection...`
indicates that the demo has been successfully run and the camera is no problem. Please refer to the following  link to view the image.
[https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera#getting-and-setting-the-app](https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera#getting-and-setting-the-app)

closed 2021-04-22 by NateDKing