ArduCAM/Arduino #502: Arducam DISPLAY GARBAGE WITH OV2640_MINI_2MP_PLUS WHEN CONNECTED TO MicroSD CARD ADAPTER

t1jani, 2020-10-12:
I have wired up my Arducam OV2640_MINI_2MP_PLUS to an the MicroSD Card Adapter HW-125 which is connected an arduino uno. I am using default example from the library included and all I get is garbage data. Here is an image of my serial monitor

[image]

ArducamSupport, 2020-10-13:
Hi，
Due to the poor design of some SD card slots, the SPI bus is not released, which cause the bus can't connect multi device. So I advise you change another SD card slot to test again.

t1jani, 2020-10-21:
When you say SPI bus do you mean the pin slot on the Arduino

closed 2020-10-26 by t1jani