ArduCAM/Arduino #504: adafruit huzzah esp8266 feather with OV5647 mini camera

Huss-coder, 2020-10-29:
Hello,
Can anyone plz tell me what sketch should I use to connect the OV5647 mini Camera with adafruit huzzah esp8266 feather?
How can I make it live video stream to my phone or labtop?
I tried to use the following sketch "Arducam_ESP8266_Filecapture" but I cant find in memorysaver the mini OV5647 to activate.
How can I add the mini OV5647 to the header data "memorysaver"????????
Thanks in advance.

ArducamSupport, 2020-11-02:
Hi，
We currently do not support connecting ov5647 on esp8266.
If you want to use arducam mini 5mp plus on esp8266, you can refer to this [repository](https://github.com/ArduCAM/ArduCAM_ESP8266_UNO).

Huss-coder, 2020-11-02:
can I use the adafruit huzzah esp8266 feather or should it be the nano type or uno boeard?