ArduCAM/Arduino #505: Arducam Rev C+ shield with OV2640 no display and SPI interface error

simpossiblee, 2020-11-01:
I am working with arduino mega 2560, arducam rev c+ shield and OV2640 camera. 
[image]
[image]

I attached both together by stacking right on top each other, with reference to arduino mega pin 8 to arducam pin 8 as shown below.
[image]

the code i used was from arducam library examples for ArduCAM_REVC_Camera_Playback and i have edited the memorysaver to uncomment #define ARDUCAM_SHIELD_REVC and #define OV2640_CAM. there is no error with the code when uploading but the result was a white screen and serial monitor showed SPI interface error.

Does anyone know what is the issue and kindly help me with this?

ArducamSupport, 2020-11-02:
Hi，
There is no problem with your example, it seems that there is a problem with your hardware.
Please check whether there is any problem with the SPI signal connection, or you can try to connect a SPI device with a wire to confirm that the SPI is normal.
[image]

simpossiblee, 2020-11-02:
@ArducamSupport Thanks for your help, i tried reconnecting the arducam and it works now. however, when i tried taking pictures, they are all corrupted shown below:
[image]

and i would like to know if even i have a shield, can i still use the host v2 app to capture photos and tried ticking 'SaveImage' but no image was saved into the SD card nor the file link. 
[image]

Is it also possible to use the IP address to view it? it prompted Request fail!

ArducamSupport, 2020-11-03:
Hi，
Is the picture above your LCD display? It seems that you are displaying the jpeg image directly on the LCD. jpeg is a compressed image and cannot be displayed directly on lcd. You can try to save the image to the SD card and then view it. I suggest you refer to this link: [https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/REVC/ArduCAM_REVC_Capture2SD](https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/REVC/ArduCAM_REVC_Capture2SD)
If you want to use the host v2 app to view images, you need to use the serial port to send the jpeg data you get to the PC. I suggest you refer to this link: [https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/REVC/ArduCAM_REVC_Video_Streaming](https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/REVC/ArduCAM_REVC_Video_Streaming)
The function of using IP address access is designed for ESP related boards, and your board does not support it.

simpossiblee, 2020-11-03:
The image above is the picture saved in the SD card when using ArduCAM_REVC_Camera_Playback exmaple. the LCD display is working fine 
[image] 
but only the image saved is corrupted. it is saved as a BMP file. 
after that i tried using ArduCAM_REVC_Capture2SD but there is no images saved at all and the serial monitor only displayed 'arducam start!' and nothing else. the LCD display is also showing white screen.

Also when i tried using ArduCAM_REVC_digital_camera, ArduCAM_REVC_Video2SD and ArduCAM_REVC_Video_streaming, the LCD display all appeared as shown below. 
[image]
 
@ArducamSupport

ArducamSupport, 2020-11-04:
Hi，
When you save the bmp image to the SD card, please make sure that the information in `bmp_header` is correct. The `bmp_header` contains the width and height information of the image you want to save. For the definition of the bmp header, please check the information yourself.
Since the LCD can display normally when using ArduCAM_REVC_Camera_Playback, you can refer to this example for use. Since this product has long been discontinued, the related library code is no longer maintained.