ArduCAM/Arduino #507: Basic Issues including Image Corruption, Data Handling, and WiFi Streaming

JoePShoulak, 2021-01-19:
Hi there! I consider myself pretty Arduino-conversant, but having gotten an Arducam has pushed that theory to its limit, so I apologize if these are overly simple questions or if there are overly-findable answers I did not see haha. 

**Image Corruption:** When I stream from my Arducam running on an Uno over USB Serial to my computer running the Host program, no matter the resolution image corruption is common, happening maybe ~1/5-1/10 times. Is there a way to try to mitigate this?

**Data Handling:** This kind of has to do with streaming, but how exactly is the picture stored before, for example, writing to SD? It looks like it's in the FIFO buffer, but I'm not entirely sure what that is. I have a screen that can display bitmap images; if I were to try to display the picture from the Arducam on the screen instead of writing it to an SD, how would I go about doing that?

**WiFi Streaming:** I do not have an ESP Device, I have several Unos, MKR 1010s, and one Mega. I have a WiFi shield coming for the Uno, and the MKRs have native WiFi. How do I use them to stream the camera to the host app on my PC?

I'm at work right now but should be home in about an hour or two where I can post code examples and error messages, but I've experienced all of these issues basically while working with two examples: OV2640 functions and Capture2SD.

Thanks for any and all help, I really do love interfacing with this camera! I just wish I knew anything at all about how to interface with it haha.

ArducamSupport, 2021-01-20:
Hi,
**Image Corruption**:This is usually caused by the SPI speed being too fast, or the wire is too long.
**Data Handling**：Before the image is read, it is stored in the FIFO in the form of jpeg, and you need to use SPI to read the data. We do not have an example of displaying the image on oled.
You can send the image data through wifi. For related examples, you can refer to our ESP32 example.
**WiFi Streaming**：Our current example of using wifi to transmit images is only ESP32, other boards in your hand may need to be debugged by yourself.
**Misc**:Theoretically, it can be tailored to fit your OLED, but there is no relevant example at present. If you need it, please contact sales for customization requirements.

JoePShoulak, 2021-01-20:
Thanks so much!
**Image Corruption:** I think my wire length is fine, it's a standard Dupont cable. How would I go about slowing down the SPI?
**The rest:** Thanks! I'll check into that!

ArducamSupport, 2021-01-22:
Hi，
For how to reduce the speed of SPI, you need to refer to the official Arduino example on SPI.