ArduCAM/Arduino #513: ArduCam OV5642 works with Arduino DUE's Wire1.begin() I2C bus line but not main Wire.begin() I2C bus

Aero2020, 2021-03-01:
Hello, I haven't been able to get my ArduCam to work with my Arduino DUE's SDA0 and SCL0 I2C bus which for example, should be made active in the "ArduCAM_Mini_5MP_OV5642_Plus_Functions" example code by replacing Wire1.begin() in line 41 with Wire.begin(). I can get the SDA1 and SCL1 I2C bus to work (by keeping Wire1.begin() ) but I don't understand why I can't use SDA0 and SCL0 on my Arduino DUE by changing it to Wire.begin(). Is there something else I need to change somewhere to make this work?

Note: In order to use the DUE with this camera I had to go to the memorysaver.h file and uncomment "#define OV5642_MINI_5MP_BIT_ROTATION_FIXED" as shown:

[image]

Any help will be greatly appreciated as I cannot use SDA1 and SDA0 lines for specific reasons. I am using Arduino IDE 1.8.12

ArducamSupport, 2021-03-27:
Hi,
The camera can run on secondary I2C, indicating that there is no problem with the camera itself. As for why the camera cannot be used on the default I2C pins, I think you should first use a simple default I2C example to test whether the default I2C is turned on correctly. Finally, connect the camera to test.

Aero2020, 2021-03-27:
Thank you for the response. I have tested an IMU with the default I2C bus and the bus works. I have also tried testing the camera with multiple Arduino DUEs and their I2C buses but only their second I2C bus works (SDA1/SCL1) and not the default bus (SDA0/SCL0) in all of them. I have come to the conclusion that it appears that the ArduCam software is what is not allowing the default I2C bus to work.

Are there any recommendations regarding the software aside from what was mentioned above? Thank you in advance.

ArducamSupport, 2021-04-07:
Sorry for keeping you waiting. After our test, if the arducam library is not modified, DUE does not support Wire0. Due to the compatibility of some boards before, DUE is defined in the Arducam library to use Wire1. If you want to use Wire0, please follow the steps below:
1. Comment out the following in Arducam.cpp.
[image]

2. Modify Wire1 to Wire in the test example
[image]

3. Connect SDA and SCL to 20 and 21 respectively

gmardiansyah, 2024-01-28:
Hi, I try connect ArduCAM on feather M0, and I try using SERCOM pin for the I2C. But, It doesn't work. Any reference for this problem ?