ArduCAM/Arduino #525: JPG decode fails with ArduCAM 2MP coz xdensity and ydensity is 0 in jpeg buffer

pkjains, 2021-06-21:
Background:
1. I connected 2MP Plus ArduCAM to Arduino Nano BLE as per the photo in this issue https://github.com/ArduCAM/Arduino/issues/442

2. I installed ArduCAM library in the Arduino IDE.
3. I uncommented  #define OV2640_MINI_2MP_PLUS in memorysaver.h
4. I am using ArduCAM_Mini_2MP_Plus_functions.ino as an example.
5. SPI interface get's detected reliably. OV2640 gets detected reliably as well.
  See the log output:
  23:13:45.876 -> ACK CMD ArduCAM Start! END
  23:13:46.075 -> ACK CMD SPI interface OK. END
  23:13:46.075 -> ACK CMD OV2640 detected. END
  23:13:46.075 -> ACK CMD switch to OV2640_320x240 END
6. In the read_fifo_burst function(), I try and print the temp value which is essentially the bytes being read from ArduCAM buffer. 

  23:14:23.472 -> temp is D8
  23:14:23.472 -> temp_last is FF
  23:14:23.472 -> temp is D8
  23:14:23.472 -> ACK IMG END
  23:14:23.472 -> ⸮⸮temp is FF
  23:14:23.472 -> ⸮temp is E0
  23:14:23.472 -> ⸮temp is 0
  23:14:23.472 ->  temp is 10
  23:14:23.472 -> temp is 4A
  23:14:23.472 -> Jtemp is 46
  23:14:23.472 -> Ftemp is 49
  23:14:23.472 -> Itemp is 46
  23:14:23.472 -> Ftemp is 0
  23:14:23.472 ->  temp is 1
  23:14:23.472 -> temp is 2
  23:14:23.472 -> temp is 2
  23:14:23.472 -> temp is 0
  23:14:23.472 ->  temp is 0
  23:14:23.506 ->  temp is 0
  23:14:23.506 ->  temp is 0
  23:14:23.506 ->  temp is 0
  23:14:23.506 ->  temp is 1
  23:14:23.506 -> temp is FF
  23:14:23.506 -> ⸮temp is B6
  23:14:23.506 -> ⸮temp is 0
  23:14:23.506 ->  temp is 86
  23:14:23.506 -> ⸮temp is 0
  ...
  ...
  ...

Problem statement:
The byte stream looks like 
FF D8     FF E0 0 10 4A 46 49 46 0    1          2 2       0 0           0 0         0 1 FF B6 ...
---------   ---------  ----- -------------------  --------    ----   -------------  -------------
Header  APP0    len     JFIF ident.     density   ver  X Density   Y Density 

According to JPEG wikipedia link, Xdensity and Ydensity cannot be ZERO. 

This is causing the jpeg decoder to fail in decoding the jpeg stream from ArduCAM.

Has anyone run into this issue or have any idea how to resolve it?

ArducamSupport, 2021-06-23:
Hi，

From your description, it is obvious that there is a problem with the data of the JPEG image.
Without modifying our example, there are usually the following possible reasons.
1. The wire is too long, resulting in unstable data transmission
2. The SPI speed is too fast, and there is an error in the transmission data.

pkjains, 2021-06-23:
I tried both 921600 (which is default in the code) and also 115200. I don't see any difference in the result. Below is how I changed it. let me know if this is ok or are there some other settings required to change the baud rate.

  #if defined(__SAM3X8E__)
    Wire1.begin();
    Serial.begin(115200);
  #else
    Wire.begin();
    Serial.begin(115200); --> Changed the value in this line. 
  #endif

pkjains, 2021-06-23:
as for the wire being too long, i am using the wire that came with the arducam via a breadboard and the Arduino nano is directly plugged into the breadboard. please see the picture. 
[image]

ArducamSupport, 2021-06-24:
Hi，

The first thing you modify is the baud rate of the serial port, not the SPI rate.
Secondly, can you try the following figure to connect the wire directly to the BLE pin? Don't pass the breadboard.

[image]
[image]

Another point, try to replace the power supply interface instead of using the hub to supply power.

kh-quan, 2021-07-25:
I am facing a similar problem but unsure if the problem is the exact same. I am capturing faulty image data and transferring them over the usb serial. Arducam Host App not showing anything. I've emailed support for help. Hopefully theres a quick fix.