ArduCAM/Arduino #528: 5MP Plus OV5642 not connecting with Arducam Host v2

tddouglas, 2021-07-17:
I'm using an Arudino Uno Wifi Rev2 board with an Arducam 5MP Plus OV5642 Mini Module Camera Shield SPI Camera Module and have wired it up just as instructed in the uno documentation:
[image]

I've configured the `memorysaver.h` file, and I've downloaded the libraries and moved them under the `arduino/libraries` folder as instructed in [readme](https://github.com/ArduCAM/Arduino). But I am unable to get any of the example sketches working. Most recently, I attempted the `ArduCam_Mini_5MP_OV5642_Plus_Functions` sketch and I am unable to get it to communicate with the Arducam Host v2. It doesn't provide the normal boot up text and only show `ArduCam Single` when I click the capture button. 
[image]

If I change the baud rate to 115200 so the Arduino Serial Monitor can read it, I continually get `"ACK CMD SPI interface Error! END"` error messages. 
[image]

Can you provide any debug steps for getting my ArduCam to work over SPI?

tddouglas, 2022-03-15:
No I had to give up on this project because I couldn't get around this. Very sad there is non-existent support

cmcfarlin49, 2022-12-21:
> hi did you ever get this figured out

For me it was a baud rate issue. Default was 921600 and had to lower to 256000 to use camera with Arduino Nano even though Nano should have been OK up to 1 MHz. Hope that helps

Darshansgit, 2022-12-21:
so im using an uno, id have to change it to what baud rate for it to work?

Atefan, 2023-09-05:
Guys i did it! I use arduino uno and a 5MP plus arducam....What I did was modify in the ArduCAM_Mini_5MP_OV5642_Plus_Functions sketch the baud rate of the Serial.begin to 115200, bc in the code it may be 921600 so i just deleted the #if defined(__SAM3X8E__) and just left Serial.begin(115200);
        Wire.begin();
        this is how it looks now:
        
        void setup() {
      // put your setup code here, to run once:
      uint8_t vid, pid;
      uint8_t temp;
        Serial.begin(115200);
        Wire.begin();
        //Serial.begin(921600);
        
      Serial.println("ACK CMD ArduCAM Start! END");
      // set the CS as an output:
      pinMode(CS, OUTPUT);
      digitalWrite(CS, HIGH);
      ...