ArduCAM/Arduino #537: Cannot execute ArduCAM_Host_V2.0_Mac under macOS Big Sur v11.6 (Intel)

sjfheaton, 2021-10-02:
Attempting to execute ArduCAM_Host_V2.0_Mac under macOS Big Sur v11.6 (Intel) fails with the message:

You do not have permission to open the application “ArduCAM_Host_V2.0_Mac”.

Is this a codesign issue?

codesign -d reports that: code object is not signed at all

Attempting to sign the app with “codesign --force --deep --sign - /ArduCAM_Host_V2.0_Mac.app” results in the message:

unsealed contents present in the bundle root

zehaog2, 2022-08-11:
Try this: [(https://ugetfix.com/ask/how-to-fix-you-dont-have-permission-to-open-the-application-error-in-mac-big-sur/)](url)

ronaldpetty, 2023-12-05:
I am using Sonoma and also had/have trouble. I was able to get the application open.

```
* In terminal, visit - /Applications/ArduCAM_Host_V2.0_Mac.app/Contents/MacOS
* sudo su
* chmod 700 ArduCAM_Mini_QT 

sh-3.2# ./ArduCAM_Mini_QT 
QMetaObject::connectSlotsByName: Connecting slot on_comboBox_pix_currentIndexChanged() with the first of the following compatible signals: ("currentIndexChanged(int)", "currentIndexChanged(QString)")
QMetaObject::connectSlotsByName: Connecting slot on_comboBox_function_currentIndexChanged() with the first of the following compatible signals: ("currentIndexChanged(int)", "currentIndexChanged(QString)")
QMetaObject::connectSlotsByName: Connecting slot on_comboBox_type_currentIndexChanged() with the first of the following compatible signals: ("currentIndexChanged(int)", "currentIndexChanged(QString)")
libpng warning: iCCP: known incorrect sRGB profile
```

Unfortunately, it seems to connect and can click around, but no image seems to be taken.

[image]

I was able to get in working on Windows, but how wants to do that!

If any ArduCAM peeps are here, I would be interested in making this work on MacOS or even Linux. I would even like to get a CLI tool to work versus this UI trouble. Any feedback appreciated.