ArduCAM/Arduino #571: How do we run arducam ov5642 mini 5mp plus with arduino uno r3?

Adigoom, 2023-03-07:
We have tried running the mini ov5642 plus 5mp with the arduino uno r3. We wired them as instructed and the code gets uploaded swifly, but the com in the host app does not open. What could be the problem?

Adigoom, 2023-03-09:
We have tried using different com ports and the problem still won't dissolve.  The error that pops up is '' SPI interface error ''
I have done some research and it all leads me to the clock pulse speed of the master(UNO R3 board/16MHz) and the slave(arducam mini 5mp plus/8MHZ).  How do we REDUSE the uno r3 clock speed to match the Arducam mini 5mp plus.

Just maybe this is the solution to our problem.

nxtdaydelivery, 2023-03-09:
the problem is probably with your wiring, as the camera is on a 4mhz clock and that is the clock speed for the board, not the SPI interface (which is controlled by the code, and does work).

could you please send a photo of your connection?

Adigoom, 2023-03-10:
[image]
[image]
[image]

nxtdaydelivery, 2023-03-10:
the gray wire should be in pin 7.

Adigoom, 2023-03-10:
We have tried it in different pins like pins 6, 7, and 10, but it still reflects the error "SPI interface error".

nxtdaydelivery, 2023-03-10:
if you change the pin, you must make sure the pin you choose reflects the one in the code.

Also, have you made sure to select the correct chip in the code (ov5642/5mp plus), and the chip select (gray wire), and using the correct example code?