ArduCAM/Arduino #578: Selecting hardware platform

fatburg3r, 2023-08-17:
Hi, I have a arduino tiny machine learning kit. https://store.arduino.cc/products/arduino-tiny-machine-learning-kit 

This kit comes with a OV7675 camera and a custom Arduino shield. What hardware platform should I choose in the memorysaver.h?

Alexxakiode, 2023-11-20:
Hi @fatburg3r please did you get the answer to your question? I have the same question too. Thank you

Alexxakiode, 2023-11-20:
Thanks @fatburg3r for this. I have tried adding both to the library and still didn't work. It only worked when I commented these out:

// Checks that the Arducam library has been correctly configured
// #if !(defined OV2640_MINI_2MP_PLUS)
// #error Please select the hardware platform and camera module in the Arduino/libraries/ArduCAM/memorysaver.h
// #endif

But I haven't been able to apply the camera for person detection

Thanks for the lead

fatburg3r, 2023-11-20:
If you are doing person detection, have you tried running the example given in Harvard_TinyMLx?

Alexxakiode, 2023-11-20:
Yes @fatburg3r giving me this error. 
Compilation error: tensorflow/lite/c/common.h: No such file or directory
I am searching to see if I can locate the file 'common.h' so I can move it to the folder atm.
Thank you

fatburg3r, 2023-11-20:
Hmm, that's weird, it's in the Harvard_TinyMLx library https://github.com/tinyMLx/arduino-library/blob/11ec7399c91af5126294f0440816c32c0dcb214c/src/tensorflow/lite/c/common.h#L4

You can try manually adding the library after downloading from github.

Alexxakiode, 2023-11-20:
Thank you @fatburg3r 

It was in the library but because I had Arduino_TensorFlowLite-1.15.0-ALPHA also, it was seeing it as multiple libraries. I have moved the Arduino_TensorFlowLite-1.15.0-ALPHA temporarily as it has been the one working for other examples. It compiled but I still can't get the camera to work:

Person score: 107 No person score: -107
Failed to initialize camera!
Image capture failed.

Is there anything else I need in addition to Shield, Arduino nano ble..., OV7675 camera please?

Thanks @fatburg3r you are such as saviour at the right time

fatburg3r, 2023-11-20:
Maybe you can switch on verbose (can be found in the preference, if I remember correctly) when compiling the script, check if the correct library in the correct directory is being compiled. If that doesn't work, I don't think I have any more suggestions that can help, sorry. 

As for the hardware, I believe the ones you mentioned are everything you need.

Alexxakiode, 2023-11-20:
@fatburg3r it seems to be working now:
Person score: 58 No person score: -58

Thank you so much!

closed 2023-12-26 by fatburg3r