ArduCAM/Arduino #590: error: 'regtype' does not name a type

FlorentNGN, 2024-05-21:
I recently bought an arducam mini 2MP, I connect it to my arduino uno R4 WIFI, I want to run the example code "ArduCAM_Mini_2MP_OV2640_functions", but when I want to compile, it says: 
C:\Users\Acer\AppData\Local\Arduino15\libraries\ArduCAM/ArduCAM.h:768:2: error: 'regtype' does not name a type; did you mean 'wctype'?
  regtype *P_CS;
  ^~~~~~~
  wctype
C:\Users\Acer\AppData\Local\Arduino15\libraries\ArduCAM/ArduCAM.h:769:2: error: 'regsize' does not name a type
  regsize B_CS;
  ^~~~~~~
C:\Users\Acer\AppData\Local\Arduino15\libraries\ArduCAM\examples\mini\ArduCAM_Mini_2MP_OV2640_functions\ArduCAM_Mini_2MP_OV2640_functions.ino:28:11: error: conflicting declaration 'const int CS'
 const int CS = 7;
           ^~
In file included from C:\Users\Acer\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.1.0\cores\arduino/Arduino.h:124:0,
                 from C:\Users\Acer\AppData\Local\Temp\arduino\sketches\D1EF11006A86C86A1F374DBB6FF436D3\sketch\ArduCAM_Mini_2MP_OV2640_functions.ino.cpp:1:
C:\Users\Acer\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.1.0\variants\UNOWIFIR4/pins_arduino.h:136:22: note: previous declaration as 'const uint8_t CS'
 static const uint8_t CS  =  PIN_SPI_CS;
                      ^~

I don’t understand why, people to help me?

keeeal, 2024-09-28:
Hey @FlorentNGN,

I am not sure if you are still working on this or managed to get it working, but I am in the same situation (Arduino Uno R4 WIFI, OV5642 instead of your OV2640) and I can explain these errors.

The output you have provided actually describes two different errors with the same root cause.

### Problem 1. `regtype` and `regsize` are not defined.

These normally get defined in `ArduCAM.h` depending on the compiler option that specifies the chip used by your Arduino.

```
#define regtype volatile uint32_t
#define regsize uint32_t
```

For example, for AVR chips [here](https://github.com/ArduCAM/Arduino/blob/066a7ea1ccae4813e9a9b9d04e55f96bb8b4039e/ArduCAM/ArduCAM.h#L120-L121) or for SAM3X8E [here](https://github.com/ArduCAM/Arduino/blob/066a7ea1ccae4813e9a9b9d04e55f96bb8b4039e/ArduCAM/ArduCAM.h#L138-L139), etc etc. The problem is that the Uno R4 WIFI doesn't use a chip in any of the currently specified families. It uses the Renesas RA4M1, for which there is currently no compiler option. The R4 Minima uses this chip too.

Solutions?
- Maybe one of the other chip's settings are correct for the chip we have but I don't know. I will try some stuff out and come back to this issue if it works. At this stage I don't even know the right way of adding an extra compiler option in the Arduino IDE yet :joy: 
- The real solution here would be to work out which compiler option is set by the Arduino compiler to specify a Renesas chip. Then I could add a new def block in the `ArduCAM.h` file with the right settings in it.

### Problem 2. `CS` is defined twice.

This one is simpler. Arudino provides some constants for each board in a file called `pins_arduino.h` and for the Uno R4 this includes a constant called CS.

```
static const uint8_t CS  =  PIN_SPI_CS;
```

The file is maintained [here](https://github.com/arduino/ArduinoCore-renesas/blob/7b8aba7282762ce78cad10f6080e08db30f4d3aa/variants/UNOWIFIR4/pins_arduino.h#L138) and for me it was installed at `/.arduino15/packages/arduino/hardware/renesas_uno/1.2.1/variants/UNOWIFIR4/pins_arduino.h`.

Unfortunately the examples in this repo also define a constant called CS. In the OV2640 example it is defined [here](https://github.com/ArduCAM/Arduino/blob/066a7ea1ccae4813e9a9b9d04e55f96bb8b4039e/ArduCAM/examples/mini/ArduCAM_Mini_2MP_OV2640_functions/ArduCAM_Mini_2MP_OV2640_functions.ino#L28) and in the OV5642 it is [here](https://github.com/ArduCAM/Arduino/blob/066a7ea1ccae4813e9a9b9d04e55f96bb8b4039e/ArduCAM/examples/mini/ArduCAM_Mini_5MP_OV5642_Plus_Functions/ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino#L30).

```
const int CS = 7;
```

Honestly, it feels like Arduino are in the wrong here... Why do they need to define a two-letter constant when `PIN_SPI_CS` already exists? And if my understanding is correct, the Chip Select pin is not Arduino board specific - It can be any output pin that can be set to high or low and you require more that one of them if you intend to select between more than one SPI device.

Solutions?
- Rename the CS variable in the example. It gets passed into ArduCAM class initializer a few lines later anyway and never used again.

### Conclusion

The root cause:

**The Arduino Uno R4 chip is not supported by ArduCAM yet**

... but it might be easy to add support.

If I get something working I will add a PR to this repo, at which point other R4 users should be able to use my fix, whether the PR get accepted or not.

keeeal, 2024-09-28:
Okay, so I went on an adventure.

It looks like the compiler options used to specify the R4 Minima and R4 WIFI are [`ARDUINO_UNOR4_MINIMA`](https://github.com/arduino/ArduinoCore-renesas/blob/7b8aba7282762ce78cad10f6080e08db30f4d3aa/boards.txt#L69C41-L69C61) and [`ARDUINO_UNOR4_WIFI`](https://github.com/arduino/ArduinoCore-renesas/blob/7b8aba7282762ce78cad10f6080e08db30f4d3aa/boards.txt#L122C73-L122C91) respectively.

I discovered that the register size of the Renesas RA4M1 is 16-bit even though it is ostensibly a 32-bit CPU. I also suspect that none of the definitions apart from `regtype` and `regsize` matter at all - I don't think any of them are used in this repo.

Finally, I made [a PR to add support for Arduino Uno R4 boards](https://github.com/ArduCAM/Arduino/pull/595).

## For future travelers:

If you are using an Arduino Uno R4 and the above PR is not merged, go to [my fork of this repo](https://github.com/keeeal/ArduCAM-Arduino-Uno-R4) and use that instead.

Good luck!

cozywolfy77, 2025-12-26:
it doesnt work, when i try it it jsut errors and is stupid
also im trying to make a uno r4 wifi with a tiny monocrome oled display with a camera that is a adrucam 2mp
and make it to where you can take pics but they are ditherd and small